contiki-ng/contiki-ng #798: Xauth key for X11 forwarding with OSX+Docker

simonduq, 2018-12-17:
In the wiki, the `contiker` aliases for OSX adds an xauth key with:
```
XAUTH_HEXKEY=`xauth list | head -n 1 | awk '{print $3}'`
````

This statically uses the first xauth entry (`head -n 1`) -- do we have any reason to believe this corresponds to the server we're binding to? On my system, I needed to use a different xauth entry, with host `smbp.local/unix:0`. To get X11 forwarding working I had to replace `head -n 1` with `grep smbp.local/unix:0`. I'm not sure how general my workaround is though...

yatch, 2018-12-17:
> do we have any reason to believe this corresponds to the server we're binding to? 

I don't think so...

It may be better to `grep` with the host name of the docker server. 

For example, add `` -e DOCKER_HOSTNAME=`hostname` `` to `docker run` command in `contiker` function and call `xauth list | grep ${DOCKER_HOSTNAME} | awk '{print $3}'` instead of using `head -n 1`?

g-oikonomou, 2019-03-30:
So I don't have an `smbp.local/unix:0` in my system, I do have various other `unix:N` entries though. Manually setting the xauth key to a randomly-selected one of those seems to have worked here.

I am trying to come up with a way to add this properly to my function in `.bash_profile`.

simonduq, 2019-03-30:
Sounds great thanks! :)

g-oikonomou, 2019-03-30:
`grep $DOCKER_HOSTNAME` alone won't work I don't think. I have two entries in the xauth list that match this. Should probably be something like `$DOCKER_HOSTNAME/unix` I think

g-oikonomou, 2019-03-30:
`        XAUTH_HEXKEY=`xauth list | grep ${DOCKER_HOSTNAME}/unix | awk '{print $3}'``

I'm pretty sure I am being stupid here, what am I missing?

PkLearning, 2019-04-07:
How to fix "Can't connect to X11" for MacOS

Check it out! 
[Dockerized UI Apps in Docker for Desktop MacOS 2018+](https://stackoverflow.com/a/53548183)

And now I can run gui with XQuartz

`alias contiker="docker run --privileged --mount type=bind,source=$CNG_PATH,destination=/home/user/contiki-ng -e DISPLAY=docker.for.mac.host.internal:0 -v /tmp/.X11-unix:/tmp/.X11-unix -v /var/run/docker.sock:/var/run/docker.sock -ti contiker/contiki-ng"
`

g-oikonomou, 2019-04-16:
> alias contiker="docker run --privileged --mount type=bind,source=$CNG_PATH,destination=/home/user/contiki-ng -e DISPLAY=docker.for.mac.host.internal:0 -v /tmp/.X11-unix:/tmp/.X11-unix -v /var/run/docker.sock:/var/run/docker.sock -ti contiker/contiki-ng" 

This actually works very nicely indeed...

g-oikonomou, 2019-10-18:
@yatch @simonduq should we change the wiki to use this `DISPLAY=docker.for.mac.host.internal:0` business? I think it will allow us to get rid of `COMMAND_STRING` from the contiker function? And perhaps convert to using an alias?

g-oikonomou, 2019-10-19:
This seems to work fine (with XQuartz):

`socat TCP-LISTEN:6000,reuseaddr,fork UNIX-CLIENT:\"$DISPLAY\"`

And then in `~/.bash_profile`

```
export CNG_PATH=<absolute-path-to-your-contiki-ng>

alias contiker="docker run --privileged \
  --mount type=bind,source=$CNG_PATH,destination=/home/user/contiki-ng \
  -e DISPLAY=docker.for.mac.host.internal:0 \
  -ti contiker/contiki-ng"
```

g-oikonomou, 2020-04-15:
@simonduq @yatch Guys shall we update the wiki as per the most recent comment and close this one?

yatch, 2020-04-17:
@g-oikonomou I will update the wiki with your lastest solution. Then, I will ask a student to set up his Mac following the wiki.

yatch, 2020-04-21:
Wiki is updated: https://github.com/contiki-ng/contiki-ng/wiki/Docker#with-xquartz :+1:

closed 2020-04-21 by yatch

Telespielstube, 2020-10-24:
Seems I still cannot get Cooja GUI it to run on my Mac. Im on the latest version 10.15.7 (19H2)

My setup looks like this:

`./bashrc:`

`export CNG_PATH=/Users/marta/contiki-ng`

`alias contiker="docker run                                                           \`
`               --privileged                                                          \`
`              --sysctl net.ipv6.conf.all.disable_ipv6=0                             \`
`               --mount type=bind,source=$CNG_PATH,destination=/home/user/contiki-ng  \`
` 	        -e DISPLAY=docker.for.mac.host.internal:0                             \`
`                -ti contiker/contiki-ng"`

I open `xQuartz` and map the tcp port 6000

`$ open -a XQuartz
$ socat TCP-LISTEN:6000,reuseaddr,fork UNIX-CLIENT:\"$DISPLAY\"`

I check the connection in a second terminal:
`$ lsof -i TCP:6000` 

`COMMAND   PID  USER   FD   TYPE             DEVICE SIZE/OFF NODE NAME`
`X11.bin 10178 marta   12u  IPv6 0xb2295e400fec2efd      0t0  TCP *:6000 (LISTEN)`
`socat   11304 marta    5u  IPv4 0xb2295e401ef72a4d      0t0  TCP *:6000 (LISTEN)`

I run Cooja from the container:

`$ contiker cooja`

and this is the output from the docker container:

`run:`
     `[java] Exception in thread "main" java.awt.AWTError: Can't connect to X11 window server using 'docker.for.mac.host.internal:0' as the value of the DISPLAY variable.`
     `[java] 	at sun.awt.X11GraphicsEnvironment.initDisplay(Native Method)`
     `[java] 	at sun.awt.X11GraphicsEnvironment.access$200(X11GraphicsEnvironment.java:65)`
     `[java] 	at sun.awt.X11GraphicsEnvironment$1.run(X11GraphicsEnvironment.java:115)`
     `[java] 	at java.security.AccessController.doPrivileged(Native Method)`
     `[java] 	at sun.awt.X11GraphicsEnvironment.<clinit>(X11GraphicsEnvironment.java:74)`
     `[java] 	at java.lang.Class.forName0(Native Method)`
     `[java] 	at java.lang.Class.forName(Class.java:264)`
     `[java] 	at java.awt.GraphicsEnvironment.createGE(GraphicsEnvironment.java:103)`
     `[java] 	at java.awt.GraphicsEnvironment.getLocalGraphicsEnvironment(GraphicsEnvironment.java:82)`
     `[java] 	at sun.awt.X11.XToolkit.<clinit>(XToolkit.java:132)`
     `[java] 	at java.lang.Class.forName0(Native Method)`
     `[java] 	at java.lang.Class.forName(Class.java:264)`
     `[java] 	at java.awt.Toolkit$2.run(Toolkit.java:860)`
     `[java] 	at java.awt.Toolkit$2.run(Toolkit.java:855)`
     `[java] 	at java.security.AccessController.doPrivileged(Native Method)`
     `[java] 	at java.awt.Toolkit.getDefaultToolkit(Toolkit.java:854)`
     `[java] 	at sun.swing.SwingUtilities2.getSystemMnemonicKeyMask(SwingUtilities2.java:2032)`
     `[java] 	at javax.swing.plaf.basic.BasicLookAndFeel.initComponentDefaults(BasicLookAndFeel.java:1158)`
     `[java] 	at javax.swing.plaf.metal.MetalLookAndFeel.initComponentDefaults(MetalLookAndFeel.java:431)`
     `[java] 	at javax.swing.plaf.basic.BasicLookAndFeel.getDefaults(BasicLookAndFeel.java:148)`
     `[java] 	at javax.swing.plaf.metal.MetalLookAndFeel.getDefaults(MetalLookAndFeel.java:1577)`
     `[java] 	at javax.swing.UIManager.setLookAndFeel(UIManager.java:539)`
     `[java] 	at javax.swing.UIManager.setLookAndFeel(UIManager.java:579)`
     `[java] 	at javax.swing.UIManager.initializeDefaultLAF(UIManager.java:1349)`
     `[java] 	at javax.swing.UIManager.initialize(UIManager.java:1459)`
     `[java] 	at javax.swing.UIManager.maybeInitialize(UIManager.java:1426)`
     `[java] 	at javax.swing.UIManager.getInstalledLookAndFeels(UIManager.java:419)`
     `[java] 	at org.contikios.cooja.Cooja.setLookAndFeel(Cooja.java:1239)`
     `[java] 	at org.contikios.cooja.Cooja.main(Cooja.java:3168)`
     `[java] Java Result: 1`

If I check socat I get the following error:

`2020/10/24 16:53:19 socat[11326] E connect(5, LEN=68 AF=1 "/private/tmp/com.apple.launchd.dJsE2ZqFeo/org.macosforge.xquartz:0", 68): Connection refused`