contiki-ng/contiki-ng #799: TSCH packet size issue

atiselsts, 2018-12-17:
TSCH can add a few bytes to a packet when adding the security header. This can cause a packet to become oversized. At the moment the radio drivers do not always handle oversized packets well, for example the CC26xx driver accepts the packet and returns the `RADIO_TX_OK` code from the `transmit()` function.

There is a security problem in here. Consider this scenario:
   - node X joins the network
   - node X sends slightly oversized packets destined to some other node through node A
   - A tries but fails to forward the packet, but does not detect that, as the radio driver returns TX_OK
   - A does not get an ACK for the packet it never sent.
   - node X repeats sending the oversized packets with high rate
   - A thinks it's link to the next hop is bad. A selects alternative routing path. The same problem repeats with the alternative nexthop.
   - A drops from the RPL network as it thinks all nexthops are bad.

Node X can be malicious or honest but misconfigured node.

There is a Cooja script and config files here to repeat the problem.
https://github.com/atiselsts/contiki-ng/blob/bugfix/packet-sizes/examples/rpl-udp/rpl-udp-cooja.csc

The various values of `UDP_PAYLOAD_SIZE` in the `project-conf.h` file correspond to different behaviors.

simonduq, 2019-04-25:
So, I think what we want is to fix the MAC header len estimate for TSCH. The faulty function is:
https://github.com/contiki-ng/contiki-ng/blame/develop/os/net/mac/tsch/tsch.c#L1160

as it calls `NETSTACK_FRAMER.length()` without setting the security parameters first.
So what we'd want in principle is to set the required packetbuf attributes exactly as in:
https://github.com/contiki-ng/contiki-ng/blame/develop/os/net/mac/tsch/tsch.c#L1046

*but*

I'm confused now because it looks like the framer function used to compute the header length:
https://github.com/contiki-ng/contiki-ng/blob/develop/os/net/mac/framer/frame802154.c#L358

includes the security auxiliary header len but not the MIC len!
And if that's correct then we also have a bug in CSMA.
Am I missing something there?

atiselsts, 2019-09-30:
The security problem is now fixed with #931.

The functionality problem remains in this corner case: silent failure in forwarding when fragmentation is disabled. But it's probably just a drawback of the 6LoWPAN specification, rather than something Contiki-NG specific. Closing the issue.

closed 2019-09-30 by atiselsts