contiki-ng/contiki-ng #800: Maximal packet size unification

atiselsts, 2018-12-17:
At the moment there are a large number of configuration settings that try to guess the maximal payload size the radio driver can transmit (which is 125 bytes for IEEE 802.15.4 2.4 GHz radios):

* NETSTACK_RADIO_MAX_PAYLOAD_LEN
* SICSLOWPAN_MAC_MAX_PAYLOAD
* TSCH_PACKET_MAX_LEN
* CSMA_MAC_LEN

Some of them are used at places, some (`SICSLOWPAN_MAC_MAX_PAYLOAD`) are never used. These should be unified; ideally, the radio driver should tell the max packet size it can handle. However, this must be a `#define` since buffer sizes depend on this.
Some radio drivers use `127` or other constants directly; some radio drivers define their own constants for max packet sizes (which may be different for Rx and Tx, for example if Rx includes the FCS but Tx does not).

Also, the radio drivers should have a unified semantics when the upper layers try to send a packet that is too big. My proposal is to validate the size at the start of `transmit()`; do not accept 0-size or oversized packets (return `RADIO_TX_ERR`).

I'm not 100% sure how the packetbuf size is related to this. It's set to 128 at the moment.

atiselsts, 2018-12-18:
Are underscore #define names acceptable for constants? Then we could do this in a clean way without changing much code - allow each radio driver to define its own limits and make the other `#defines` dependent on that:

```
/* New code in  cc2538 radio driver's header */
#define cc2538_rf_driver_max_payload_len 125

/* Existing code in cc2538 platform's header */
#define NETSTACK_RADIO cc2538_rf_driver

/* In the netstack.h header */
#define NETSTACK_RADIO_MAX_PAYLOAD_LEN_XX(radio) radio##_max_payload_len
#define NETSTACK_RADIO_MAX_PAYLOAD_LEN_X(radio) NETSTACK_RADIO_MAX_PAYLOAD_LEN_XX(radio)
#define NETSTACK_RADIO_MAX_PAYLOAD_LEN NETSTACK_RADIO_MAX_PAYLOAD_LEN_X(NETSTACK_RADIO)

/* In a TSCH header */
#define TSCH_PACKET_MAX_LEN MIN(NETSTACK_RADIO_MAX_PAYLOAD_LEN, PACKETBUF_SIZE)

```

debug-ito, 2018-12-20:
Pull-request #558 removed sicslowpan's `MAC_MAX_PAYLOAD`, but `SICSLOWPAN_CONF_MAC_MAX_PAYLOAD` remains in os/contiki-default-conf.h etc. We should remove that macro.

`NETSTACK_RADIO_MAX_PAYLOAD_LEN` is currently defined by some platforms, but maybe it's a good idea to make it the canonical place of configuration.

atiselsts, 2019-01-11:
Any thoughts?

simonduq, 2019-01-11:
No strong opinion on the proposal above, but I very much support the idea of homogenizing this!

atiselsts, 2019-09-27:
Fixed by #908 and #974

closed 2019-09-27 by atiselsts