contiki-ng/contiki-ng #807: cc1310 + cc1190

marcomilazzo, 2018-12-27:
Hi
has anybody implemented the the cc1190 code for this PA ?
ciao marco

rafaelchrist, 2019-04-02:
Hi marcoilazzo, did you do any development with CC1310+CC1190? I'm looking to port contiki-NG to LAUNCHXL-CC13-90. Thanks.

marcomilazzo, 2019-04-02:
hi
i'm working on it
at present time i'm working on antennas .they can make a lot of differance in range
i'm going in some days to test contiki with well tuned antennas and see the range
the porting should be easy maybe !!

 ti_lib_ioc_pin_type_gpio_input(CC1190_HGM);
    ti_lib_ioc_io_port_pull_set(CC1190_HGM, IOC_IOPULL_UP);
    ti_lib_ioc_port_configure_set(CC1190_LNA, IOC_PORT_RFC_GPO0, IOC_STD_OUTPUT | IOC_PORT_GPIO);
    ti_lib_ioc_port_configure_set(CC1190_PA, IOC_PORT_RFC_GPO1, IOC_STD_OUTPUT | IOC_PORT_GPIO);

if you do it please contact me
ciao

rafaelchrist, 2019-04-10:
Hi, sorry for the delayed answer. I worked on it and I was able to make contiki-ng to work on Texas LAUNCHXL-CC13-90.

marcomilazzo, 2019-04-10:
hi
no problem :)
in some days i will try range with a collinear antenna ( if it works )
these antennas have a 5db increase . i got some equipment to tune it
can you tell me what you have done ? where you added the code  and ..........
if you are interested in antennas i can send you all the info
marco

mdmobashir, 2020-03-02:
Hi,
We have been working on it and it seems that antenna does make a lot of difference. Currently, the background noise flood seems to increase from -120 to -100 for 1130-1190. 
Any help on that, please?

joakimeriksson, 2026-10-21:
Closing as a question. There is no CC1190 or LAUNCHXL-CC1310-90 board support in Contiki-NG today. If you have a working port (PA/LNA pin mapping via RFC GPO), please send it as a board PR under arch/platform/simplelink. Please use GitHub Discussions for usage questions.

closed 2026-10-21 by joakimeriksson