contiki-ng/contiki-ng #808: Can I run border-router.native on RPI with different slip-radio simultaneously?

Rockabilly15, 2019-01-04:


g-oikonomou, 2019-01-17:
Currently we don't support a single border router with more than one radio. You can of course run two border routers and each will create a separate network - you will need to make sure the two networks use different prefixes otherwise you are likely to run into problems.

The device or resource busy seems irrelevant. This will happen if a process tries to open a serial port that has already been opened by a different process

closed 2019-01-17 by g-oikonomou

160221J, 2021-01-11:
I am trying to run two border routers (of two separate networks) in same Raspberry Pi (RPi 4). What are the prefixes that I need to modify in order to run both simultaneously? 
I have modified the PAN ID of the two slip radios that I have uploaded for both cc2650. What are the other parameters need to modify in slip radio? I also got the message 'buzy' while trying to run this.