contiki-ng/contiki-ng #809: ble ipv6 example can not work.

desword, 2019-01-06:
I am working on the ble-ipv6 example in https://github.com/contiki-ng/contiki-ng/tree/develop/examples/platform-specific/cc26x0-cc13x0/ble-ipv6

However, there is only ble-ipv6 client example, but no ble-ipv6 server example. 

So I copy the code from https://github.com/spoerk/contiki/tree/master/examples/cc26xx/cc26xx-ble-server-demo.

And make the following changes, the changes are anotated.
The project_conf.h
```
#ifndef PROJECT_CONF_H_
#define PROJECT_CONF_H_
#define MODE_IEEE802_15_4 1   /* enable RPL and disable TSCH in Makefile */
#define MODE_BLE      2

#define MODE MODE_BLE
/*---------------------------------------------------------------------------*/
/* Disable button shutdown functionality */
#define BUTTON_SENSOR_CONF_ENABLE_SHUTDOWN    0
/*---------------------------------------------------------------------------*/
/* Change to match your configuration */
#define BOARD_CONF_DEBUGGER_DEVPACK           1
/*---------------------------------------------------------------------------*/
#define PACKETBUF_CONF_SIZE                   1280
#define QUEUEBUF_CONF_NUM                       1
#define UIP_CONF_BUFFER_SIZE                  1280

#define CC26XX_CONF_RADIO_MODE          CC26XX_RADIO_MODE_BLE
#define NETSTACK_CONF_RADIO                 ble_cc2650_driver
//#define NETSTACK_CONF_RDC                   ble_null_par_driver
//#define NETSTACK_CONF_MAC                   ble_l2cap_driver

#define RTIMER_CONF_MULTIPLE_ACCESS       1

/* BLE radio settings */
#define BLE_MODE_CONF_INIT_PEER_ADDR      0x2471891A2A82

/* BLE L2CAP settings */
#define BLE_CONF_DEVICE_NAME          "TI CC26xx server device"

/* 6LoWPAN settings */
//#define SICSLOWPAN_CONF_MAC_MAX_PAYLOAD       1280
//#define SICSLOWPAN_CONF_COMPRESSION           SICSLOWPAN_COMPRESSION_HC06
//#define SICSLOWPAN_CONF_COMPRESSION_THRESHOLD   0  /* always use compression */
//#define SICSLOWPAN_CONF_FRAG                    0
//#define SICSLOWPAN_FRAMER_HDRLEN                0

/*/ * 6LoWPAN settings * / */
#define SICSLOWPAN_CONF_MAC_MAX_PAYLOAD       1280
#define SICSLOWPAN_CONF_COMPRESSION           SICSLOWPAN_COMPRESSION_6LORH
#define SICSLOWPAN_CONF_COMPRESSION_THRESHOLD   0  /* always use compression */
#define SICSLOWPAN_CONF_FRAG                    0
#define SICSLOWPAN_FRAMER_HDRLEN                0

/* network layer settings */
// #define UIP_CONF_ROUTER                         1
// #define UIP_CONF_ND6_SEND_NA                    1
// #define UIP_CONF_ND6_SEND_RA          1
// #define UIP_CONF_IP_FORWARD                     0
// #define UIP_CONF_ND6_SEND_NS                    1

/*/ * network stack settings * / */
#define UIP_CONF_ROUTER                         0
#define UIP_CONF_ND6_SEND_NA                1

/*---------------------------------------------------------------------------*/
#endif /* PROJECT_CONF_H_ */
```

Note that I also remove the `ble_null_par_driver`, because when I enable it, it returns the compile errors:
```
  CC        server.c
  LD        server.elf
obj_cc26x0-cc13x0/ble-hal-cc26xx.o: In function `process_thread_ble_hal_conn_rx_process':
ble-hal-cc26xx.c:(.text.process_thread_ble_hal_conn_rx_process+0x348): undefined reference to `ble_l2cap_driver'
obj_cc26x0-cc13x0/rf-core.o: In function `process_thread_rf_core_process':
rf-core.c:(.text.process_thread_rf_core_process+0x54): undefined reference to `ble_l2cap_driver'
obj_cc26x0-cc13x0/contiki-main.o: In function `main':
contiki-main.c:(.text.startup.main+0x1e0): undefined reference to `ble_l2cap_driver'
obj_cc26x0-cc13x0/netstack.o: In function `netstack_init':
netstack.c:(.text.netstack_init+0x1c): undefined reference to `ble_l2cap_driver'
obj_cc26x0-cc13x0/sicslowpan.o: In function `output':
sicslowpan.c:(.text.output+0x488): undefined reference to `ble_l2cap_driver'
collect2: error: ld returned 1 exit status
make: *** [server.elf] Error 1
```

In the server.c, the only changes are the removed redefinition of UIP_IP_BUF    
```
//#define UIP_IP_BUF    ((struct uip_ip_hdr *)&uip_buf[UIP_LLH_LEN])
```

After all that. I can compile the server successfully. However, it can not detect the client and create connections.

The output from the server are like :
```
[INFO: Main      ] Starting Contiki-NG-develop/v4.2-50-g2b0c2d5-dirty
[INFO: Main      ] - Routing: RPL Lite
[INFO: Main      ] - Net: sicslowpan
[INFO: Main      ] - MAC: CSMA
[INFO: Main      ] - 802.15.4 PANID: 0xabcd
[INFO: Main      ] - 802.15.4 Default channel: 26
[INFO: Main      ] Node ID: 44931
[INFO: Main      ] Link-layer address: 0012.4b00.0d5e.af83
[INFO: Main      ] Tentative link-local IPv6 address: fe80::212:4b00:d5e:af83
[INFO: CC26xx/CC13xx] TI CC2650 LaunchPad
[INFO: CC26xx/CC13xx]  RF: Channel 0
CC26XX-IPv6-over-BLE server started
[INFO: Main      ] Starting Contiki-NG-develop/v4.2-50-g2b0c2d5-dirty
[INFO: Main      ] - Routing: RPL Lite
[INFO: Main      ] - Net: sicslowpan
[INFO: Main      ] - MAC: CSMA
[INFO: Main      ] - 802.15.4 PANID: 0xabcd
[INFO: Main      ] - 802.15.4 Default channel: 26
[INFO: Main      ] Node ID: 44931
[INFO: Main      ] Link-layer address: 0012.4b00.0d5e.af83
[INFO: Main      ] Tentative link-local IPv6 address: fe80::212:4b00:d5e:af83
[INFO: CC26xx/CC13xx] TI CC2650 LaunchPad
[INFO: CC26xx/CC13xx]  RF: Channel 0
CC26XX-IPv6-over-BLE server started
```

**Anyone can give some hints?**

desword, 2019-01-06:
I have also tried one possible direction by modifying the network configuration in project_conf.h of server to 
```
/* network layer settings */
 #define UIP_CONF_ROUTER                         1
 #define UIP_CONF_ND6_SEND_NA                    1
 #define UIP_CONF_ND6_SEND_RA          1
 #define UIP_CONF_IP_FORWARD                     0
 #define UIP_CONF_ND6_SEND_NS                    1
```
Note that I add the new definitions `UIP_CONF_ND6_SEND_NS`. Without this, it returns the error of `defrt is defined but not used`. However, the result is only reducing the frequency of showing the output or the scanning period.

desword, 2019-01-14:
I have just fixed it.
We should add all of the UIP_CONF_ROUTER related codes into the `net/ble-l2cap.c` and `rf-core/ble-hal/ble-hal-cc26xx.c` following the code from `https://github.com/spoerk/contiki/tree/master/cpu/cc26xx-cc13xx`.

And of cause, the PEER_ADDR in server and SERVER_IP in client.c should be configured correctly

closed 2019-01-14 by desword

sponchiado, 2019-07-24:
@desword is it possible to run and simulate ble ipv6 example with Cooja? I´m getting an error because of the redefinition of NETSTACK_CONF_RADIO. It is defined as cooja_radio_driver inside Cooja files and in project-conf.h from de BLE IPV6 example it is defined as ble_cc2650_driver. Did you get this problem when trying to use Cooja? Thanks in advance!

NourDjidi, 2022-04-25:
Hello @[sponchiado](https://github.com/sponchiado) and @[desword](https://github.com/desword), could you please tell me if you were able to simulate ble ipv6 example with cooja ? Thanks in advance