contiki-ng/contiki-ng #810: Using a mote equiped with higher amount of RAM and flash memory in Cooja

h-hosseini, 2019-01-06:
We (Dr. @ElisaRojas and @davidcawork) set up Contiki-NG by the Native toolchain installation (Linux). We have only Installed ARM (for ARM-based platforms such as CC2538DK and Zoul) and MSP430 (for MSP430 and MSPSim emulator in Cooja) compilers. Since the amount of RAM and flash memory in the sky mote is low for our application, we want to use another mote type in Cooja. When we create every mote type  in Cooja (except the Sky mote), compiling the new mote is encountered an error such as "... Makefile.include:92: *** The target platform "..." doesn't exist (maybe it was misspelled?).  stop. Process returned error code 2". Note that the platform of the motes in Cooja(except the Sky mote) are not in the Contiki-NG platform folder, and any platform (except the Sky platform) have not a mote type in Cooja in our setting. How can we use another mote type? Thanks for your response!

h-hosseini, 2019-01-12:
We install Contiki-NG by the Docker image ([This shell script](https://get.docker.com/) installs Docker on a deb/rpm based Linux platform if the platform is supported by Docker), and it seems that some of motes have not developed in Contiki-NG so far. Anyway, is there any way to run RPL (both storing and non-storing mode) with the (all) logs on the Sky mote? For example, is there any unessential module to disable/remove it before compiling?

g-oikonomou, 2019-01-17:
I would suggest trying native Cooja motes to see if they will meet your requirements. #819 proposes adding back the Z1 platform, which is also supported in Cooja and has higher capability than sky motes, but it still relatively limited compared to cooja motes.

closed 2019-01-17 by g-oikonomou