contiki-ng/contiki-ng #826: BLE on Simplelink platform not working

firmwareguru, 2019-01-17:
I've tried to get the BLE beacon functionality going with the new simplelink platform and it's not going well.  The example app locks up in the RF_pendCmd() in sched.c on the first beacon broadcast attempt.

I have "ported" the `platform-specific/cc26x0-cc13x0/cc26x0-demo.c` example to build with the simplelink platform - this just required renaming a couple of defines.

I have tried this on the launchpad/cc1350 as well as a cc1352-based board - same result.  I can build the cc26x0-demo example for the cc26x0-cc13x0 target for the same launchpad/cc1350 board and it works fine - beacons are captured by my phone.

@tiepettersen I am wondering if you can help out here?  

First step is to get BLE beacon daemon going with the simplelink target.  Next I'd like to get the ble-hal ported over to simplelink as well, but first things first.

tiepettersen, 2019-01-17:
@firmwareguru, sounds strange. I'll take a look.

tiepettersen, 2019-01-17:
Ok, I found out what the issue is. The CFE RF Patch is wrong, as a special CPE patch is required when running multi-protocol. Somehow the correct patches did not propagate when pushing for a PR. I'm developing a fix now. I'll also test that BLE beaconing works as well.

firmwareguru, 2019-01-17:
Super, thanks.

tiepettersen, 2019-01-18:
So I'm not able to push out a PR before the weekend, but I have developed a fix for CC13x2/CC26x2 devices. However, there seems to be an underlying problem of running multi-protocol on CC13x0/CC26x0 devices that I will need to work out. Aiming for a PR next monday.

firmwareguru, 2019-01-18:
I look forward to trying it out.

tiepettersen, 2019-01-21:
Fixed by #829

tiepettersen, 2019-01-29:
This can now be closed.

closed 2019-01-29 by g-oikonomou