contiki-ng/contiki-ng #827: No debug information output When `sudo ./border-router.native fd00::1/64 -s /dev/ttyACM0`

desword, 2019-01-17:
I want to debug the program of slip-radio on CC2650 with BLE and the rpl-border-router with native TARGET on RPI.

However, when I execute the command: `sudo ./border-router.native fd00::1/64 -s /dev/ttyACM0`, all of the debug information belonging to the rpl-border-router is shown, but none of the slip-radio.
The output log is:
```
[INFO: Main      ] Starting Contiki-NG-develop/v4.2-50-g2b0c2d5e4-dirty
[INFO: Main      ] - Routing: RPL Lite
[INFO: Main      ] - Net: sicslowpan
[INFO: Main      ] - MAC: br-mac
[INFO: Main      ] - 802.15.4 PANID: 0xabcd
[INFO: Main      ] - 802.15.4 Default channel: 26
[INFO: Main      ] Node ID: 1800
[INFO: Main      ] Link-layer address: 0102.0304.0506.0708
[INFO: IPv6 DS   ] Init: 300 neighbors
[INFO: IPv6 DS   ] 2 default routers
[INFO: IPv6 DS   ] 3 prefixes
[INFO: IPv6 DS   ] 0 routes
[INFO: IPv6 DS   ] 3 unicast addresses
[INFO: IPv6 DS   ] 5 multicast addresses
[INFO: IPv6 DS   ] 2 anycast addresses
[INFO: IPv6 DS   ] Adding prefix fe80::length 64, flags 0, Valid lifetime 0, Preffered lifetime 0
[INFO: RPL       ] initializing
[INFO: Main      ] Tentative link-local IPv6 address: fe80::302:304:506:708
[INFO: Main      ] Platform init 
[INFO: Native    ] Added global IPv6 address fd00::302:304:506:708
[INFO: IPv6 Route] Add default
[INFO: IPv6 Route] Add default: adding default route to fd00::1
[INFO: Main      ] Platform inti finished 
[DBG : Main      ] With RPL Border Router
[INFO: RPL BR    ] Contiki-NG Border Router started
[INFO: BR        ] RPL-Border router started
ifconfig tun0 inet `hostname` up
ifconfig tun0 add fd00::1/64
ifconfig tun0

tun0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet 127.0.1.1  netmask 255.255.255.255  destination 127.0.1.1
        inet6 fe80::92da:b295:b7f8:d014  prefixlen 64  scopeid 0x20<link>
        inet6 fd00::1  prefixlen 64  scopeid 0x0<global>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

[INFO: TCP/IP    ] input: received 48 bytes
[INFO: IPv6      ] packet received from fe80::92da:b295:b7f8:d014 to ff02::2
[INFO: IPv6      ] icmpv6 input length 48 type: 133 
[DBG : IPv6      ] Upper layer checksum len: 8 from: 40
[ERR : IPv6      ] Unknown ICMPv6 message type/code 133
[INFO: IPv6 DS   ] Init: 300 neighbors
[INFO: IPv6 DS   ] 2 default routers
[INFO: IPv6 DS   ] 3 prefixes
[INFO: IPv6 DS   ] 0 routes
[INFO: IPv6 DS   ] 3 unicast addresses
[INFO: IPv6 DS   ] 5 multicast addresses
[INFO: IPv6 DS   ] 2 anycast addresses
[INFO: IPv6 DS   ] Adding prefix fe80::1:ff06:708length 64, flags 0, Valid lifetime 0, Preffered lifetime 0
[INFO: RPL       ] initializing
[INFO: BR        ] Setting prefix fd00::1
[INFO: RPL       ] IPv6 addresses:
[INFO: RPL       ] -- fd00::ce78:abff:fe77:4b03
[INFO: RPL       ] -- fe80::ce78:abff:fe77:4b03
[INFO: RPL       ] reset DIO timer (Init root)
[INFO: RPL       ] created DAG with instance ID 0, DAG ID fd00::ce78:abff:fe77:4b03, rank 128
[INFO: RPL       ] created a new RPL DAG
[INFO: BR        ] Server IPv6 addresses:
[INFO: BR        ]   fd00::ce78:abff:fe77:4b03
[INFO: BR        ]   fe80::ce78:abff:fe77:4b03

-------------Start slip radio procedure

[INFO: RPL       ] sending a multicast-DIO with rank 128 to ff02::1a
[DBG : IPv6      ] Upper layer checksum len: 76 from: 40
[INFO: ICMPv6    ] Sending ICMPv6 packet to ff02::1a, type 155, code 1, len 72
[INFO: TCP/IP    ] output: sending to (NULL LL addr)
[INFO: 6LoWPAN   ] output: sending IPv6 packet with len 116
[DBG : 6LoWPAN   ] compression: before (76): 60000000004c3a40fe80000000000000ce78abfffe774b03ff02000000000000000000000000001a9b017d4100f0008008f00000fd00000000000000ce78abfffe774b03040e00080c00040000800001001e003c081e4040ffffffffffffffff00000000fd000000000000000000000000000000
[DBG : 6LoWPAN   ] compression: first header: 58
[DBG : 6LoWPAN   ] compression: last header could is not compressed: 58
[DBG : 6LoWPAN   ] compression: after (4): 7a3b3a1a
[INFO: 6LoWPAN   ] output: header len 40 -> 4, total len 116 -> 80, MAC max payload 110, frag_needed 0
[INFO: Frame 15.4] Out:  1 ffff.0000.0000.0000 15 80 (95)

```

Actually, there should be logs belonging to the slip-radio after `[INFO: Frame 15.4] Out:  1 ffff.0000.0000.0000 15 80 (95)`.  

For comparison, I also plot the slimilar log that output from CC2538. And the command is `sudo ./border-router.native fd00::1/64 -s /dev/ttyUSB0`. I only plot the related log start from `-------------Start slip radio procedure`.

```
[INFO: RPL       ] sending a multicast-DIO with rank 128 to ff02::1a
[DBG : IPv6      ] Upper layer checksum len: 76 from: 40
[INFO: ICMPv6    ] Sending ICMPv6 packet to ff02::1a, type 155, code 1, len 72
[INFO: TCP/IP    ] output: sending to (NULL LL addr)
[INFO: 6LoWPAN   ] output: sending IPv6 packet with len 116
[DBG : 6LoWPAN   ] compression: before (76): 60000000004c3a40fe8000000000000002124b0005af80daff02000000000000000000000000001a9b015df100f0008008f00000fd0000000000000002124b0005af80da040e00080c00040000800001001e003c081e4040ffffffffffffffff00000000fd000000000000000000000000000000
[DBG : 6LoWPAN   ] compression: first header: 58
[DBG : 6LoWPAN   ] compression: last header could is not compressed: 58
[DBG : 6LoWPAN   ] compression: after (4): 7a3b3a1a
[INFO: 6LoWPAN   ] output: header len 40 -> 4, total len 116 -> 80, MAC max payload 110, frag_needed 0
[INFO: Frame 15.4] Out:  1 ffff.0000.0000.0000 15 80 (95)
[DBG : slip-radio] SR-SIN: 108 '!S'
[DBG : slip-radio] sending 1 (95 bytes)
[INFO: CSMA      ] sending to 0000.0000.0000.0000, len 95, seqno 133, queue length 1, free packets 7
[DBG : CSMA      ] scheduling transmission in 0 ticks, NB=0, BE=3
[INFO: CSMA      ] preparing packet for 0000.0000.0000.0000, seqno 133, tx 0, queue 1
[INFO: CSMA      ] tx to 0000.0000.0000.0000, seqno 133, status 0, tx 0, coll 0
[INFO: CSMA      ] packet sent to 0000.0000.0000.0000, seqno 133, status 0, tx 1, coll 0
[DBG : CSMA      ] free_queued_packet, queue length 0, free packets 8
[DBG : slip-radio] Slip-radio: packet sent! sid: 1, status: 0, tx: 1
```

I can not capture the logs from ttyACM0 on the RPI when executing the command: ` sudo ./border-router.native fd00::1/64 -s /dev/ttyACM0`, because the port is occupied.

Anyone can help me ?Thanks!

desword, 2019-01-17:
Note that I can monitor the output from /dev/ttyACM0 without executing the command `sudo ./border-router.native fd00::1/64 -s /dev/ttyACM0`.

THe output like:
```
[DBG : IPv6      ] Upper layer checksum len: 40 from: 40
[INFO: IPv6 NDP  ] Sending RA to ff02::1 from fe80::ce78:abff:fe77:4b03
[INFO: IPv6 DS   ] Sending periodic RA
[DBG : IPv6 DS   ] Random time 1 = 368
[DBG : IPv6 DS   ] Random time 2 = 16
[DBG : IPv6 DS   ] Random time 3 = 16
[INFO: TCP/IP    ] output: sending to (NULL LL addr)

[DBG : IPv6      ] Upper layer checksum len: 40 from: 40
[INFO: IPv6 NDP  ] Sending RA to ff02::1 from fe80::ce78:abff:fe77:4b03
[INFO: IPv6 DS   ] Sending periodic RA
[DBG : IPv6 DS   ] Random time 1 = 329
[DBG : IPv6 DS   ] Random time 3 = 329
[INFO: TCP/IP    ] output: sending to (NULL LL addr)

-------------after I plugin the CC2650 with mqtt-client.
[INFO: BLE-RADIO ] initiator_rx() entry finished
[INFO: BLE-RADIO ] BLE-HAL: connection (0x0000) created
[DBG : L2CAP     ] [ L2CAP ] ble-mac: input: sending L2CAP conn_req: cid: 65, addr:  24:71:89:FF:FE:1A:2A:82[DBG : L2CAP     ] 
[DBG : L2CAP     ] [ L2CAP ] send_l2cap_conn_req
[DBG : L2CAP     ] [ L2CAP ] send_l2cap_conn_req: CID: 65, MTU: 1280, MPS: 252, credits:  5
[INFO: BLE-RADIO ] stop initiating

[DBG : L2CAP     ] input 18 bytes
[DBG : L2CAP     ] [ L2CAP ] l2cap_conn_resp: connection established
[DBG : L2CAP     ] input 35 bytes
礼{;: [INFO: TCP/IP    ] output: processing 29 bytes packet from 900::1:224:7189:fffe:1a2a:8200 to ::1
[INFO: IPv6 Route] Looking up route for ::1
[WARN: IPv6 Route] No route found
[ERR : TCP/IP    ] output: destination off-link and no default route
[DBG : L2CAP     ] input 35 bytes
[INFO: TCP/IP    ] output: processing 29 bytes packet from 900::1:224:7189:fffe:1a2a:8200 to ::1
[INFO: IPv6 Route] Looking up route for ::1
[WARN: IPv6 Route] No route found
[ERR : TCP/IP    ] output: destination off-link and no default route

```

Note that, the mqtt-client acutally can create the connection with my slip-radio. However, I can not get the parent list. (Maybe the RPL is not support on BLE for now).
But I can not get the log output, therefore, it is hard for me debug.

g-oikonomou, 2019-01-17:
When you execute `border-router.native` it takes control of the serial port. Is there any chance you changed the SLIP setting to Debug for your CC2538 firmware, but did not for the CC2650? Otherwise, I cannot immediately see what could be wrong

closed 2019-02-08 by g-oikonomou