contiki-ng/contiki-ng #832: Platform zoul: power-mgmt module's pm_get_voltage() does not return correct measurements

lukasostendorf, 2019-01-24:
The function pm_get_voltage() in arch/platform/zoul/remote-revb/power-mgmt.c does not return correct voltage readings.

Here are the results of measuring the voltage every 10seconds:
[image]

In a branch of Zolertia's contiki repository I've [found a fix](https://github.com/Zolertia/contiki/commit/aafc740211417c9c46c1a5f1cda7a63e86bb2bf0) for this. Apparently they use the following conversion:
`voltage = -181.9*(register_val - 1294.72)`

The converted measurements are here:

[image]

I'm not sure whether this fitting is suitable for different battery types, but still the conversion seems more useful than the current result of the function. Therefore, I'd include the conversion also in contiki-ng.

I do not know whether I'm allowed to "copy" the commit (and refer to it in my PR?). That's why I did not come up with a PR immediately.

g-oikonomou, 2019-02-08:
Thanks for this. I'll flag it as a bug to make sure it doesn't fall through the cracks.

g-oikonomou, 2019-04-16:
Hi @lukasostendorf can you have a look at #902 if you get a chance?

PR #902 (2019-05-04): Fix zoul battery voltage conversion and add example
As pointed out in #832, the zoul battery voltage conversion is incorrect.

Turns out this has been fixed already by the zolertia team [in this commit](https://github.com/Zolertia/contiki/commit/aafc740211417c9c46c1a5f1cda7a63e86bb2bf0). The same commit adds a test example.

This pull resolves some conflicts in this commit and brings it into our codebase. We also do some minor code style and naming convention fixes.

Fixes #832
closed 2019-05-04 by g-oikonomou