contiki-ng/contiki-ng #833: Cooja mote: Transmission with radio turned off puts it back on.

pciolkosz, 2019-01-28:
https://github.com/contiki-ng/contiki-ng/blob/17f4c688880df3b242036dc121ce215795a11911/arch/platform/cooja/dev/cooja-radio.c#L214

Turning radio off in Cooja mote is not blocking outgoing packets. Even if it is a design choice, it does it wrong, because sometimes it allows to receive packet if there is incoming transmission around that time.

yatch, 2019-01-28:
Indeed, seems it should return `RADIO_TX_ERR` there. I checked the history of the file but I cannot find why this code was introduced: https://github.com/contiki-ng/contiki-ng/commit/bee0ea4834259b18d57ad5edb479c008cb3197c9#diff-9366018bcde46e7af8a0e01013daca30

> because sometimes it allows to receive packet if there is incoming transmission around that time.

By the way, just out of curiosity, did you see this actually happened in Cooja?

pciolkosz, 2019-02-01:
Yes, I saw confirmed transmission to a node with radio turned off and believe this was the cause.

g-oikonomou, 2019-02-08:
@yatch are we looking at a bug here?

yatch, 2019-02-11:
@g-oikonomou Yes; I cannot see any sensible explanation for that. I will send a PR later.

yatch, 2019-02-25:
As per the discussion at PR #855, the reported behavior is common in H/W radios. 

@pciolkosz 

> because sometimes it allows to receive packet if there is incoming transmission around that time.

I checked the code in cooja as well, but I cannot see how this can happen. It seems the radio won't receive anything during transmission. If you can give more information, I would see further. I'm closing this issue for now.

simonduq, 2019-02-25:
So I'm not so sure about the internals of cooja motes on the java side, but, maybe, what happens is:
1. we set `simRadioHWOn`
2. we call `cooja_mt_yield`
3. Cooja will run one tick and call us back after that. Perhaps it will accept any incoming packet during that tick?
4. we unset `simRadioHWOn`

In that case what we need is probably a cooja fix :)
Or maybe we don't even need to set `simRadioHWOn` in order to transmit?

yatch, 2019-02-25:
To me, It seems the radio doesn't accept an incoming frame during transmission.

https://github.com/contiki-ng/cooja/blob/63538bbb882ba06a7b8cf97c11ce2fe4d22e4f88/java/org/contikios/cooja/contikimote/interfaces/ContikiRadio.java#L183

I'd like to have a sample code to reproduce the issue...

pciolkosz, 2019-02-28:
Setup that I used was 1 node with rpl-udp example server code and 100 client nodes. Server node code was modified to call NETSTACK_MAC.off() on button click, clicked by Cooja controlling script.
First issue I observed was that sometimes this call did not turn the radio off at all. It was usually if node was transmitting around the time of the call. I changed the code to call off function 10 times with small delay as a work around, but observed, that there were a small chance to still receive some transmission by the server node.

I found log file that shows such behavior. At timestamp 600008000 and a few next timestamps node tried to turn off the radio:
600008000:1:[INFO: App       ] Trying to turn the radio off again
It seems it succeeds, because there is a number of transmissions failing after that point:
600155792:10:[INFO: CSMA      ] tx to 0001.0001.0001.0001, seqno 99, status 2, tx 0, coll 0
600647792:21:[INFO: CSMA      ] tx to 0001.0001.0001.0001, seqno 140, status 2, tx 0, coll 1
601535792:3:[INFO: CSMA      ] tx to 0001.0001.0001.0001, seqno 74, status 2, tx 4, coll 0
and so on.
Then, node with the radio shutdown sends DIO message:
639438000:1:[INFO: RPL       ] sending a multicast-DIO with rank 128 to ff02::1a
639439000:1:[INFO: CSMA      ] sending to 0000.0000.0000.0000, len 27, seqno 43, queue length 2, free packets 62
639479808:1:[INFO: CSMA      ] packet sent to 0000.0000.0000.0000, seqno 42, status 0, tx 1, coll 0
639518344:1:[INFO: CSMA      ] packet sent to 0000.0000.0000.0000, seqno 43, status 0, tx 1, coll 0
Now for some reason it receives a packet:
640104040:1:[INFO: CSMA      ] received packet from 0002.0002.0002.0002, seqno 170, len 71
node sending the packet also considers transmission successful:
640104040:2:[INFO: CSMA      ] packet sent to 0001.0001.0001.0001, seqno 170, status 0, tx 1, coll 0

Revisiting this case, it seems the radio on node 1 went back on. Previously I thought It received only a few packets, but now I found out it was active until the end of the simulation.
Here is the whole log file. It runs a bit modified version of RPL.
[log.zip](https://github.com/contiki-ng/contiki-ng/files/2915173/log.zip)

yatch, 2019-02-28:
The radio will stay turned on after a transmission in the following scenario:
1. initial state: the radio is on
1. the radio starts a transmission (entering `radio_send()` of `cooja-radio.c`)
1. the radio state is saved for later use (`radiostate` in `radio_send()` is `1`)
1. during the transmission, the "radio off" button is clicked
1. the radio is turned off; the transmission is interrupted (cancelled)
1. the radio is get back on using `radiostate`

I think your "radio off" button implementation needs to see the radio state before calling `NETSTACK_RADIO.off()` or `NETSTACK_MAC.off()`

pciolkosz, 2019-02-28:
It does not fix issue with it turning back on if the transmission fails, see edit of my previous post.
It is also not supported to check radio status, see:
https://github.com/contiki-ng/contiki-ng/blob/17f4c688880df3b242036dc121ce215795a11911/arch/platform/cooja/dev/cooja-radio.c#L306

yatch, 2019-02-28:
> It does not fix issue with it turning back on if the transmission fails

Sorry for the confusion. I didn't tell a solution; I explained an expected behavior of the current code with the scenario where you clicked the button during a transmission.

Anyway, according to your today's post, the following is not correct, is it?
> Turning radio off in Cooja mote is not blocking outgoing packets. Even if it is a design choice, it does it wrong, because sometimes it allows to receive packet if there is incoming transmission around that time.

pciolkosz, 2019-02-28:
> Anyway, according to your today's post, the following is not correct, is it?
> 
> > it allows to receive packet if there is incoming transmission around that time.
Yes, it actually turns the radio on permanently if the transmission fails.

yatch, 2019-02-28:
I believe, if the radio is off when entering `radio_send()`, the radio will get off after the transmission regardless of transmission success or failure...

> Yes, it actually turns the radio on permanently if the transmission fails.

To my understandings, this is not true.

As a side note, I also think `radio_on()` and other functions should be smarter. For instance, `radio_on()` should return an error (`-1`) if it called during transmission, that is, while the radio is busy.

pciolkosz, 2019-02-28:
Isn't it true that if any of the errors in lines between
https://github.com/contiki-ng/contiki-ng/blob/17f4c688880df3b242036dc121ce215795a11911/arch/platform/cooja/dev/cooja-radio.c#L216
and 
https://github.com/contiki-ng/contiki-ng/blob/17f4c688880df3b242036dc121ce215795a11911/arch/platform/cooja/dev/cooja-radio.c#L231
happens, line
https://github.com/contiki-ng/contiki-ng/blob/17f4c688880df3b242036dc121ce215795a11911/arch/platform/cooja/dev/cooja-radio.c#L242
wont get executed and the radio will stay on?

yatch, 2019-02-28:
Aha, I see. Now I get your point. Thanks.

yatch, 2019-02-28:
I will open a separate ticket describing the issue concisely, which will have a reference to this issue.

closed 2019-02-28 by yatch

pciolkosz, 2019-03-05:
Would it be possible to add a compilation flag to disable turning the radio on for outgoing transmission? 
It would be useful to be able to simulate node power outage under Cooja and unless I am mistaken turning the radio off is the only reliable way.