contiki-ng/contiki-ng #834: Delaying the startup of particular node by some time.

abhiverma866, 2019-01-29:
I want to delay one of the nodes in my network such that it sends its first DIS after 600 seconds. 
by default it is #define RPL_DIS_START_DELAY 5, however the node sends its first DIS earlier than this time. 
for 
#define RPL_DIS_START_DELAY 30 
first DIS is sent at 7 second from the start of node
#define RPL_DIS_START_DELAY             60
first DIS is sent at 37 second from the start of node
however further increasing doesn't work. 
What can be the possible solution ?

firmwareguru, 2019-01-31:
Indeed, the handling RPL DIS packets is in need of closer inspection.

We see that 'next_dis' is computed as follows in rpl-timers.c:
```
  next_dis = RPL_DIS_INTERVAL / 2 +
    ((uint32_t)RPL_DIS_INTERVAL * (uint32_t)random_rand()) / RANDOM_RAND_MAX -
    RPL_DIS_START_DELAY;
```

But this is confusing.  With random_rand() potentially throwing up any number between 0 and 65535, the results could be much smaller than the divisor RANDOM_RAND_MAX.  Furthermore, the START_DELAY is subtracted??

The use of `next_dis` is as if it is a second-interval counter such that any start delays that make the value greater than RPL_DIS_INTERVAL (60) will just trigger an immediate DIS transmission.

```
  next_dis++;
  if(dag == NULL && next_dis >= RPL_DIS_INTERVAL) {
    next_dis = 0;
    dis_output(NULL);
  }
```

I could propose that next_dis is a decrementing counter that when counted down to 0 would dispatch the DIS.  and next_dis is initialized to a larger number that represents RPL_DIS_INTERVAL + START_DELAY plus some randomness thrown in.

yatch, 2019-02-01:
@firmwareguru yes, I'm confused as well :-)

To make sure every mote has `RPL_DIS_START_DELAY` before the first DIS, `rpl_reset_periodic_timer()` should be:
```
void
rpl_reset_periodic_timer(void)
{
  next_dis = RPL_DIS_INTERVAL / 2 +
    ((uint32_t)RPL_DIS_INTERVAL / 2 * (uint32_t)random_rand()) / RANDOM_RAND_MAX -
    RPL_DIS_START_DELAY;
  ctimer_set(&periodic_timer, CLOCK_SECOND, handle_periodic_timer, NULL);
}
```
Put '/ 2' after the second `RPL_DIS_INTERVAL`.

@abhiverma866 Anyway, time from the mote boot to the first DIS has randomness as @firmwareguru pointed out. And, `RPL_DIS_START_DELAY` shouldn't be greater than `RPL_DIS_INTERVAL / 2`. So, 

> What can be the possible solution ?

I think you need to modify the code so that a particular mote has delay of 600 seconds exact, like:

* change the type of `next_dis` to a signed one
* replace the assignment of `next_dis` in `rpl_reset_periodic_timer()` with `next_dis = RPL_DIS_INTERVAL - RPL_DIS_START_DELAY`
* define `RPL_CONF_DIS_START_DELAY` with `600`

Hope this helps.

abhiverma866, 2019-02-01:
I found other way to delay the mote DIS. I just used stimer.

PROCESS_THREAD(sensor_process, ev, data) {
  
  PROCESS_BEGIN();
  **stimer_set(&stimer_timer, 60);
    while(stimer_expired(&stimer_timer)!=1)
			{          
          PRINTF("Waiting for timer to expire \n");
			}**
 
  PRINTF("Starting UDP server \n");

abhiverma866, 2019-02-01:
yes.. and thanks for the valuable suggestion.

closed 2019-02-01 by abhiverma866

firmwareguru, 2019-02-01:
There is still a problem with the DIS delay in standard RPL - as implemented it doesn't work.  RPL-lite is better implemented in terms of DIS timer but doesn't have a configurable startup delay parameter.
 
But I guess the creator of this issue found a work around for the problem at hand and so this issue can be closed.