contiki-ng/contiki-ng #844: Subscribe don't work in example mqtt-client.c

EdenMarcosBragadeOliveira, 2019-02-13:
Hello,

I'mj trying subscribe a topic using mqtt-client.c embedded in CC2650, according to "https://github.com/contiki-ng/contiki-ng/wiki/Tutorial:-MQTT", but it does'nt work.

I think the problem is the #define MQTT_CLIENT_ORG_ID "quickstart". When i debug the system output is: "[DBG : mqtt-client] Using 'quickstart': Skipping subscribe".

Can anybody help me?

Thanks.

g-oikonomou, 2019-02-13:
Quickstart does not allow subscriptions, so disabling subscribe when `ORG_ID` is "quickstart" is intentional. If you are using the IBM platform, you are stuck without subscriptions. If you are using your own MQTT broker, just change `ORG_ID`

closed 2019-02-13 by g-oikonomou

EdenMarcosBragadeOliveira, 2019-02-13:
Hello George,

Now it's running. I'm using the Mosquitto Broker. Then i've change the ORG_ID.

Thank you very much!