contiki-ng/contiki-ng #849: CC1352P1 rtimer problem

rnascunha, 2019-02-19:
Hi,

I'm doing the tutorials with the simplelink plataform, for CC1352p1 Lauchpad. Running the timers tutorials, the board keeps rebooting. I comment the lines (82 and 83) that set the **rtimer** at the example:

```
//rtimer_set(&timer_rtimer, RTIMER_NOW() + RTIMER_SECOND / 2, 0,
//             rtimer_callback, NULL);
```

After that, the example is working fine, but without the **rtimer**:
```
CTimer callback called
CTimer callback called
CTimer callback called
Timer process: SUCCESS
CTimer callback called
CTimer callback called
CTimer callback called
Timer process: SUCCESS

```

Is there any idea about this problem? Is that issue only at this example or at any place I try to use the **rtimer**?

Thanks!

rnascunha, 2019-02-19:
Maybe this can help:

```
using saved target 'simplelink'
----------------- Make variables: --------------
##### "TARGET": ________________________________ simplelink
##### "BOARD": _________________________________ launchpad/cc1352p1
##### "MAKE_MAC": ______________________________ MAKE_MAC_CSMA
##### "MAKE_NET": ______________________________ MAKE_NET_IPV6
##### "MAKE_ROUTING": __________________________ MAKE_ROUTING_RPL_LITE
----------------- C variables: -----------------
##### "PROJECT_CONF_PATH": _____________________ ><
##### "CONTIKI_VERSION_STRING": ________________ == "Contiki-NG-release/v4.2-177-g3642dee5d-dirty"
##### "FRAME802154_CONF_VERSION":_______________ == (0x01)
##### "IEEE802154_CONF_PANID":__________________ == 0xabcd
##### "IEEE802154_CONF_DEFAULT_CHANNEL": _______ == 26
##### "QUEUEBUF_CONF_NUM": _____________________ == 8
##### "NBR_TABLE_CONF_MAX_NEIGHBORS": __________ == 16
##### "NETSTACK_MAX_ROUTE_ENTRIES": ____________ == 16
##### "UIP_CONF_BUFFER_SIZE": __________________ == 1280
##### "UIP_CONF_UDP": __________________________ == 1
##### "UIP_CONF_UDP_CONNS": ____________________ == 8
##### "UIP_CONF_TCP": __________________________ == 0
##### "UIP_CONF_TCP_CONNS": ____________________ == 0
##### "UIP_CONF_ND6_SEND_RA": __________________ == 0
##### "UIP_CONF_ND6_SEND_NS": __________________ == 0
##### "UIP_CONF_ND6_SEND_NA": __________________ == 1
##### "UIP_CONF_ND6_AUTOFILL_NBR_CACHE": _______ == 1
##### "SICSLOWPAN_CONF_FRAG": __________________ == 1
##### "SICSLOWPAN_CONF_COMPRESSION": ___________ == SICSLOWPAN_COMPRESSION_IPHC
##### "ENERGEST_CONF_ON": ______________________ == 0
##### "LOG_CONF_LEVEL_RPL": ____________________ == 0
##### "LOG_CONF_LEVEL_TCPIP": __________________ == 0
##### "LOG_CONF_LEVEL_IPV6": ___________________ == 0
##### "LOG_CONF_LEVEL_6LOWPAN": ________________ == 0
##### "LOG_CONF_LEVEL_NULLNET": ________________ == 0
##### "LOG_CONF_LEVEL_MAC": ____________________ == 0
##### "LOG_CONF_LEVEL_FRAMER": _________________ == 0
##### "LOG_CONF_LEVEL_6TOP": ___________________ == 0
##### "LOG_CONF_LEVEL_COAP": ___________________ == 0
##### "LOG_CONF_LEVEL_LWM2M": __________________ == 0
##### "LOG_CONF_LEVEL_MAIN": ___________________ == 3
------------------------------------------------
```

firmwareguru, 2019-02-19:
Hi, the problem is platform-specific.  On the Simplelink platforms you can't use printf from the rtimer callback which is run in RTC interrupt context.  Remove the printf from `rtimer_callback` and you'll be good to go. 

Similar to https://github.com/contiki-ng/contiki-ng/issues/816

rnascunha, 2019-02-19:
Thanks @firmwareguru, working perfectly now!

Maybe this could be documented, or at the example, or at the plataform-specific page.

Thanks again.

firmwareguru, 2019-02-19:
I was thinking the same thing - have this documented on a platform-specific page for simplelink.  I am not aware of any platform-specific documentation wiki pages.  A readme could be added to arch, arch/cpu/simplelink-xxxx or arch/platform/simplelink.  Would that have help you @rnascunha ?  Would you have looked there?

rnascunha, 2019-02-19:
It does have a plataform specific wiki page:
https://github.com/contiki-ng/contiki-ng/wiki/Platform-simplelink

And I did read it. Probably a README inside the arch directory would duplicate the information.

closed 2019-02-19 by rnascunha