contiki-ng/contiki-ng #850: Cooja mote creation error when using a test script

h-hosseini, 2019-02-19:
What system are you on? Are you aware of the fact that Cooja motes are only supported on 32-bit linux? See https://github.com/contiki-ng/contiki-ng/issues/134

For now the preferred option to run Cooja motes is Docker https://github.com/contiki-ng/contiki-ng/wiki/Docker

_Originally posted by @simonduq in https://github.com/contiki-ng/contiki-ng/issues/353#issuecomment-376073836_

h-hosseini, 2019-02-19:
cc @ElisaRojas and @davidcawork

We use Docker and Cooja mote in our RPL simulation. There is a test script to control the simulation. When running the simulation without GUI,  it returns an error for **some** seed numbers:

`
FATAL [main] (Cooja.java:1338) - Exception when loading simulation: 
org.contikios.cooja.Cooja$SimulationCreationException: Mote type creation error: Error when creating corecomm instance: Lib1
	at org.contikios.cooja.Cooja.loadSimulationConfig(Cooja.java:3452)
	at org.contikios.cooja.Cooja.loadSimulationConfig(Cooja.java:3362)
	at org.contikios.cooja.Cooja.quickStartSimulationConfig(Cooja.java:1331)
	at org.contikios.cooja.Cooja.main(Cooja.java:3240)
Caused by: org.contikios.cooja.MoteType$MoteTypeCreationException: Error when creating corecomm instance: Lib1
	at org.contikios.cooja.CoreComm.createCoreComm(CoreComm.java:326)
	at org.contikios.cooja.contikimote.ContikiMoteType.doInit(ContikiMoteType.java:407)
	at org.contikios.cooja.contikimote.ContikiMoteType.configureAndInit(ContikiMoteType.java:368)
	at org.contikios.cooja.contikimote.ContikiMoteType.setConfigXML(ContikiMoteType.java:1341)
	at org.contikios.cooja.Simulation.setConfigXML(Simulation.java:703)
	at org.contikios.cooja.Cooja.loadSimulationConfig(Cooja.java:3433)
	... 3 more
Caused by: java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at org.contikios.cooja.CoreComm.createCoreComm(CoreComm.java:318)
	... 8 more
Caused by: java.lang.UnsatisfiedLinkError: org.contikios.cooja.corecomm.Lib1.init()V
	at org.contikios.cooja.corecomm.Lib1.init(Native Method)
	at org.contikios.cooja.corecomm.Lib1.<init>(Lib1.java:50)
	... 13 more
`
The log includes :
`
 INFO [main] (Cooja.java:1316) - > Starting Cooja
 INFO [main] (Cooja.java:2857) - External tools default settings: /external_tools_linux.config
 INFO [main] (Cooja.java:2887) - External tools user settings: /home/user/.cooja.user.properties
 INFO [main] (Simulation.java:422) - Simulation random seed: 123467
 INFO [main] (CompileContiki.java:140) - > make sink.cooja TARGET=cooja 
Message:
Message:> make sink.cooja TARGET=cooja 
Message:  CC        sink.c
Message:ar rcf build/cooja/mtype821.a build/cooja///obj/simEnvChange.o build/cooja///obj/cooja_mt.o build/cooja///obj/cooja_mtarch.o build/cooja///obj/rtimer-arch.o build/cooja///obj/slip.o build/cooja///obj/watchdog.o build/cooja///obj/int-master.o build/cooja///obj/beep.o build/cooja///obj/button-sensor.o build/cooja///obj/ip.o build/cooja///obj/leds-arch.o build/cooja///obj/moteid.o build/cooja///obj/pir-sensor.o build/cooja///obj/rs232.o build/cooja///obj/vib-sensor.o build/cooja///obj/clock.o build/cooja///obj/cooja-log.o build/cooja///obj/cfs-cooja.o build/cooja///obj/cooja-radio.o build/cooja///obj/eeprom.o build/cooja///obj/slip-arch.o build/cooja///obj/random.o build/cooja///obj/sensors.o build/cooja///obj/leds.o build/cooja///obj/contiki-main.o build/cooja///obj/ctimer.o build/cooja///obj/log.o build/cooja///obj/autostart.o build/cooja///obj/energest.o build/cooja///obj/etimer.o build/cooja///obj/node-id.o build/cooja///obj/mutex.o build/cooja///obj/stimer.o build/cooja///obj/compower.o build/cooja///obj/process.o build/cooja///obj/rtimer.o build/cooja///obj/timer.o build/cooja///obj/stack-check.o build/cooja///obj/button-hal.o build/cooja///obj/serial-line.o build/cooja///obj/spi.o build/cooja///obj/nullradio.o build/cooja///obj/gpio-hal.o build/cooja///obj/dbl-circ-list.o build/cooja///obj/crc16.o build/cooja///obj/ifft.o build/cooja///obj/ccm-star.o build/cooja///obj/dbl-list.o build/cooja///obj/ringbufindex.o build/cooja///obj/list.o build/cooja///obj/circular-list.o build/cooja///obj/aes-128.o build/cooja///obj/trickle-timer.o build/cooja///obj/memb.o build/cooja///obj/heapmem.o build/cooja///obj/assert.o build/cooja///obj/ringbuf.o build/cooja///obj/queuebuf.o build/cooja///obj/nbr-table.o build/cooja///obj/packetbuf.o build/cooja///obj/netstack.o build/cooja///obj/linkaddr.o build/cooja///obj/net-debug.o build/cooja///obj/link-stats.o build/cooja///obj/mac-sequence.o build/cooja///obj/mac.o build/cooja///obj/frame802154e-ie.o build/cooja///obj/framer-802154.o build/cooja///obj/nullframer.o build/cooja///obj/frame802154.o build/cooja///obj/csma-security.o build/cooja///obj/anti-replay.o build/cooja///obj/csma.o build/cooja///obj/csma-output.o build/cooja///obj/ccm-star-packetbuf.o build/cooja///obj/uipbuf.o build/cooja///obj/uip-icmp6.o build/cooja///obj/uip-sr.o build/cooja///obj/uip-ds6.o build/cooja///obj/uip-packetqueue.o build/cooja///obj/tcp-socket.o build/cooja///obj/uiplib.o build/cooja///obj/tcpip.o build/cooja///obj/ip64-addr.o build/cooja///obj/uip-udp-packet.o build/cooja///obj/uip-ds6-route.o build/cooja///obj/psock.o build/cooja///obj/uip-ds6-nbr.o build/cooja///obj/uip6.o build/cooja///obj/resolv.o build/cooja///obj/uip-nameserver.o build/cooja///obj/uip-nd6.o build/cooja///obj/udp-socket.o build/cooja///obj/simple-udp.o build/cooja///obj/sicslowpan.o build/cooja///obj/rpl-icmp6.o build/cooja///obj/rpl-of0.o build/cooja///obj/rpl-dag.o build/cooja///obj/rpl-timers.o build/cooja///obj/rpl-ext-header.o build/cooja///obj/rpl.o build/cooja///obj/rpl-nbr-policy.o [... cut ...]
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at org.contikios.cooja.CoreComm.createCoreComm(CoreComm.java:318)
	... 8 more
Caused by: java.lang.UnsatisfiedLinkError: org.contikios.cooja.corecomm.Lib1.init()V
	at org.contikios.cooja.corecomm.Lib1.init(Native Method)
	at org.contikios.cooja.corecomm.Lib1.<init>(Lib1.java:50)
	... 13 more

yatch, 2019-02-22:
Can you provide a way to reproduce this issue on one of our branches, for instance, `develop` branch?

h-hosseini, 2019-02-23:
Dear @yatch,
cc @ElisaRojas and @davidcawork

We are using release 4.2: https://github.com/contiki-ng/contiki-ng/tree/release/v4.2 , and `CONTIKI_VERSION_STRING` is `"CONTIKI_VERSION_STRING": ________________ == "Contiki-NG-release/v4.2-129-g4928a5e-dirty"` according to the result of `make TARGET=cooja viewconf`.
For more detailed information, the result of `make TARGET=cooja viewconf` is :

`
----------------- Make variables: --------------
##### "TARGET": ________________________________ cooja
##### "BOARD": _________________________________ 
##### "MAKE_MAC": ______________________________ MAKE_MAC_CSMA
##### "MAKE_NET": ______________________________ MAKE_NET_IPV6
##### "MAKE_ROUTING": __________________________ MAKE_ROUTING_RPL_CLASSIC
----------------- C variables: -----------------
##### "PROJECT_CONF_PATH": _____________________ == "project-conf.h"
##### "CONTIKI_VERSION_STRING": ________________ == "Contiki-NG-release/v4.2-129-g4928a5e-dirty"
##### "FRAME802154_CONF_VERSION":_______________ == (0x01)
##### "IEEE802154_CONF_PANID":__________________ == 0xabcd
##### "IEEE802154_CONF_DEFAULT_CHANNEL": _______ == 26
##### "QUEUEBUF_CONF_NUM": _____________________ == 64
##### "NBR_TABLE_CONF_MAX_NEIGHBORS": __________ == 26
##### "NETSTACK_MAX_ROUTE_ENTRIES": ____________ == 100
##### "UIP_CONF_BUFFER_SIZE": __________________ == 1280
##### "UIP_CONF_UDP": __________________________ == 1
##### "UIP_CONF_UDP_CONNS": ____________________ == 8
##### "UIP_CONF_TCP": __________________________ == 0
##### "UIP_CONF_TCP_CONNS": ____________________ == 0
##### "UIP_CONF_ND6_SEND_RA": __________________ == 0
##### "UIP_CONF_ND6_SEND_NS": __________________ == 0
##### "UIP_CONF_ND6_SEND_NA": __________________ == 1
##### "UIP_CONF_ND6_AUTOFILL_NBR_CACHE": _______ == 1
##### "SICSLOWPAN_CONF_FRAG": __________________ == 1
##### "SICSLOWPAN_CONF_COMPRESSION": ___________ == SICSLOWPAN_COMPRESSION_IPHC
##### "ENERGEST_CONF_ON": ______________________ == 0
##### "LOG_CONF_LEVEL_RPL": ____________________ == 0
##### "LOG_CONF_LEVEL_TCPIP": __________________ == 0
##### "LOG_CONF_LEVEL_IPV6": ___________________ == 0
##### "LOG_CONF_LEVEL_6LOWPAN": ________________ == 0
##### "LOG_CONF_LEVEL_NULLNET": ________________ == 0
##### "LOG_CONF_LEVEL_MAC": ____________________ == 0
##### "LOG_CONF_LEVEL_FRAMER": _________________ == 0
##### "LOG_CONF_LEVEL_6TOP": ___________________ == 0
##### "LOG_CONF_LEVEL_COAP": ___________________ == 0
##### "LOG_CONF_LEVEL_LWM2M": __________________ == 0
##### "LOG_CONF_LEVEL_MAIN": ___________________ == 3
------------------------------------------------
`

fwcd, 2020-06-02:
Possibly related: I am getting an `UnsatisfiedLinkError` when creating a Cooja mote for [the HTTP example](https://github.com/contiki-ng/contiki-ng/blob/develop/examples/websocket/http-example.c) (even though I am using Docker and it compiles successfully):

`java.lang.UnsatisfiedLinkError: /home/user/src/build/cooja/mtype658.cooja: /home/user/src/build/cooja/mtype658.cooja: undefined symbol: tcp_attach`

<details>
<summary>Detailed Log from Cooja</summary>

```
     [java]  INFO [AWT-EventQueue-0] (ContikiMoteType.java:406) - Creating core communicator between Java class Lib1 and Contiki library '/home/user/src/build/cooja/mtype658.cooja'
     [java] FATAL [AWT-EventQueue-0] (Cooja.java:2184) - Exception when creating mote type
     [java] org.contikios.cooja.MoteType$MoteTypeCreationException: Error when creating corecomm instance: Lib1
     [java]     at org.contikios.cooja.CoreComm.createCoreComm(CoreComm.java:326)
     [java]     at org.contikios.cooja.contikimote.ContikiMoteType.doInit(ContikiMoteType.java:407)
     [java]     at org.contikios.cooja.contikimote.ContikiMoteType.configureAndInit(ContikiMoteType.java:368)
     [java]     at org.contikios.cooja.Cooja.doCreateMoteType(Cooja.java:2179)
     [java]     at org.contikios.cooja.Cooja.doCreateMoteType(Cooja.java:2158)
     [java]     at org.contikios.cooja.Cooja$GUIEventHandler.actionPerformed(Cooja.java:2939)
     [java]     at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
     [java]     at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
     [java]     at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
     [java]     at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
     [java]     at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
     [java]     at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:842)
     [java]     at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:886)
     [java]     at java.awt.Component.processMouseEvent(Component.java:6539)
     [java]     at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
     [java]     at java.awt.Component.processEvent(Component.java:6304)
     [java]     at java.awt.Container.processEvent(Container.java:2239)
     [java]     at java.awt.Component.dispatchEventImpl(Component.java:4889)
     [java]     at java.awt.Container.dispatchEventImpl(Container.java:2297)
     [java]     at java.awt.Component.dispatchEvent(Component.java:4711)
     [java]     at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
     [java]     at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
     [java]     at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
     [java]     at java.awt.Container.dispatchEventImpl(Container.java:2283)
     [java]     at java.awt.Window.dispatchEventImpl(Window.java:2746)
     [java]     at java.awt.Component.dispatchEvent(Component.java:4711)
     [java]     at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
     [java]     at java.awt.EventQueue.access$500(EventQueue.java:97)
     [java]     at java.awt.EventQueue$3.run(EventQueue.java:709)
     [java]     at java.awt.EventQueue$3.run(EventQueue.java:703)
     [java]     at java.security.AccessController.doPrivileged(Native Method)
     [java]     at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
     [java]     at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
     [java]     at java.awt.EventQueue$4.run(EventQueue.java:733)
     [java]     at java.awt.EventQueue$4.run(EventQueue.java:731)
     [java]     at java.security.AccessController.doPrivileged(Native Method)
     [java]     at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
     [java]     at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
     [java]     at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
     [java]     at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
     [java]     at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
     [java]     at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
     [java]     at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
     [java]     at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
     [java] Caused by: java.lang.reflect.InvocationTargetException
     [java]     at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
     [java]     at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
     [java]     at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
     [java]     at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
     [java]     at org.contikios.cooja.CoreComm.createCoreComm(CoreComm.java:318)
     [java]     ... 43 more
     [java] Caused by: java.lang.UnsatisfiedLinkError: /home/user/src/build/cooja/mtype658.cooja: /home/user/src/build/cooja/mtype658.cooja: undefined symbol: tcp_attach
     [java]     at java.lang.ClassLoader$NativeLibrary.load(Native Method)
     [java]     at java.lang.ClassLoader.loadLibrary0(ClassLoader.java:1941)
     [java]     at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1824)
     [java]     at java.lang.Runtime.load0(Runtime.java:809)
     [java]     at java.lang.System.load(System.java:1086)
     [java]     at org.contikios.cooja.corecomm.Lib1.<init>(Lib1.java:49)
     [java]     ... 48 more
```

</details>

closed 2020-11-25 by g-oikonomou