contiki-ng/contiki-ng #851: VcXsrv on windows requires additional setup

StaySharp0, 2019-02-20:
https://github.com/contiki-ng/contiki-ng/wiki/Docker#on-windows

```
1. Install VcXsrv v1.19.6.0 (download-vcxsrv)
2. Run VcXsrv and check 'Disable access control' box at Extra settings
3. Open PowerShell and set variable for DISPLAY  
...
4. Running your docker  
...
```

If you had not checked, an 'Can't connect to X11 window server using 'localhost:0.0' as the value of the DISPLAY variable.' error will occur during the cooja build.

Please modify if I've got something wrong here.

yatch, 2019-02-22:
I will check this later with a Windows 10 machine which I've not set up the docker environment.

yatch, 2019-04-17:
@StaySharp0 `localhost` from the view point of a container is the container itself, not the host where VcXsrv is running. Try `host.docker.internal` instead of `localhost`.

I will update the wiki after playing wit my Windows machine a little bit more.

yatch, 2019-04-18:
@StaySharp0 I updated https://github.com/contiki-ng/contiki-ng/wiki/Docker#on-windows

I tested it with my machine. Try it again!
[image]

closed 2019-04-18 by yatch