contiki-ng/contiki-ng #852: Outdated Wiki for NAT64 and not working script

kai-chi, 2019-02-21:
There are broken links in [this section](https://github.com/contiki-ng/contiki-ng/wiki/NAT64-for-Contiki%E2%80%90NG#installing-jool) to Jool.
Also the tool for ip64 `/tools/ip64/jool-start.sh` is not working correctly on my machine. Seems like `myIP` is not read correctly because of grep on an invalid string and later unexpecte tokens in Jool (see log below):

> Configuring jool for 
net.ipv4.conf.all.forwarding = 1
net.ipv6.conf.all.forwarding = 1
Unexpected token: '-4'
Available options: instance stats global pool4 bib session file 
Unexpected token: '--enable'
Available options: instance stats global pool4 bib session file 

My machine:
Ubuntu 18.10 64-bit
Jool 4.0.0.0

joakimeriksson, 2019-02-21:
Thanks for reporting. I updated the wiki - will try out the Jool script on an updated ubuntu soon, and see if we find a fix.

navan93, 2019-03-10:
I am also facing the same issue. Any update on this?.
While on this topic, is there anyway I can access the mesh nodes from another computer on the same local network as the rpl-border-router?

joakimeriksson, 2019-03-10:
I will try the same thing on 18.04 (I think I was using 16.04 when it worked) so I am hoping to set up a PR on this fairly soon.

joakimeriksson, 2019-03-17:
I just tried latest version of Jool (4.0) and it is significantly different from the old one - so we will need to completely redo the wiki and startup scripts.

nvt, 2026-09-30:
Fixed in #3132. The broken script is gone and the NAT64 docs were rewritten around the built-in NAT64.

closed 2026-09-30 by nvt