contiki-ng/contiki-ng #854: RPL classic is sending DAOs in MOP3

g-oikonomou, 2019-02-22:
With RPl Classic, I am seeing DAO messages going out even when RPL is in MOP3

`#define `RPL_CONF_MOP RPL_MOP_NO_DOWNWARD_ROUTES`

Unless I have misunderstood something in the spec, I am not expecting to see any DAO messages whatsoever in this MOP. I think we are missing a check somewhere. Will investigate further, but I thought I'd put an issue here to make sure it doesn't fall through the cracks.

I observed this with v4.2, I have not checked if it is still ocurring in develop. We can close this if it's fixed.

g-oikonomou, 2019-02-22:
In rpl-dag.c L833 (`rpl_select_dag()`) we have this:

```C
  if(best_dag->preferred_parent != last_parent) {
    rpl_set_default_route(instance, rpl_parent_get_ipaddr(best_dag->preferred_parent));
    LOG_INFO("Changed preferred parent, rank changed from %u to %u\n",
           (unsigned)old_rank, best_dag->rank);
    RPL_STAT(rpl_stats.parent_switch++);
    if(RPL_IS_STORING(instance)) {
      if(last_parent != NULL) {
        /* Send a No-Path DAO to the removed preferred parent. */
        dao_output(last_parent, RPL_ZERO_LIFETIME);
      }
      /* Trigger DAO transmission from immediate children.
       * Only for storing mode, see RFC6550 section 9.6. */
      RPL_LOLLIPOP_INCREMENT(instance->dtsn_out);
    }
    /* The DAO parent set changed - schedule a DAO transmission. */
    rpl_schedule_dao(instance);
    rpl_reset_dio_timer(instance);
    if(LOG_DBG_ENABLED) {
      rpl_print_neighbor_list();
    }
  } else if(best_dag->rank != old_rank) {
```

It looks like we are correctly checking the mode before scheduling a no-path DAO, but we are skipping the check for a DAO to what I think is the new parent.

simonduq, 2019-02-24:
Yes I think that's the issue. But ofc we don't want to just check for `RPL_IS_STORING` here, because we need the DAO also in non-storing. And there may be more code locations where DAOs are sent and a check needed.

In rpl-lite this is handled directly in `rpl_icmp6_dao_output`, which returns immediately if in mode no-downwards-routes

PR #2232 (2022-10-28): Don't send DAO when in MOP 0
Fixes #854
closed 2022-10-28 by nfi