contiki-ng/contiki-ng #859: RPL-lite node on CC1352R1 goes down at medium udp traffic

mariusns, 2019-02-28:
I modded the example of rpl-udp and set up a network with 10 nodes using CC1352R1, using simple_udp_sendto and simple_udp_register toghter with ctimers.  Each node sends health status back to root every 5 sec, except one node sending data each ~400 ms, as a simple stress test for the network.  Everything goes fine as log as the high freq node is connected to root directly. 

The radio is configured for  prop-mode sending on 868 Mhz

When having 1-2 nodes in between root, the  high frequency node together with parallel nodes connected to the same parent, goes down. Leaving the network for  RPL_DELAY_BEFORE_LEAVING before reappear and failing again after ~30 seconds.

I tried with more frequent transmissions, and the conditions only get worse, failing faster. 

For the first tries I used NETWORK_ROUTING.node_is_reachable() as a send condition, but this only put the parents node out for the same time. Then moving over to if(curr_instance.dag.preferred_parent != NULL) keept the parent alive while the child nodes went down. 

I did had a try with more 5 or 6 nodes connected to the same node, and a similar situation occurred.

project-conf.h looks like: 
#define SUPPORTS_PROP_MODE                  1
#define RF_CONF_MODE 						RF_MODE_SUB_1_GHZ
#define WATCHDOG_CONF_DISABLE				1
#define RADIO_TXPOWER_CONF					12
#define NETSTACK_CONF_RADIO					prop_mode_driver
#define IEEE802154_CONF_DEFAULT_CHANNEL     26
#define DOT_15_4G_CONF_FREQ_BAND_ID         DOT_15_4G_FREQ_BAND_863

and Makefile: 

CFLAGS += -g -O0
MAKE_ROUTING = MAKE_ROUTING_RPL_LITE

firmwareguru, 2019-03-02:
This is an interesting issue.  I have some cc1352r1 boards I can try to repro with.

mariusns, 2019-03-05:
I had a further look at this, and it looks like combination of best_parent selection and the time penalty when failing to transmit. For the first case there is the nbr_link_metric(rpl_nbr_t *nbr) which checks the etx against a rather low value of MAX_LINK_METRIC = 512, that fails.  As a result the node goes down instead of selecting a new parent, because of the other condition of rank test, in acceptable_rank.

Etx is given penalty on retransmissions.  Also  the time delay on may increase to 32/128 of a sec. This doesnt help when you only have one nbr in your rpl-neighbours-list, when having 5-6 or more sub-nodes transmitting on ~1Hz or more.