contiki-ng/contiki-ng #863: CoAP + DTLS not working

rnascunha, 2019-03-01:
Hi,

I'm trying to use CoAP with DTLS but its not working. I'm using a CC1352P1 launchpad, with the Simplelink platarfom to test. I'm compiling the `coap-example-server` with the following options:

`MAKE_WITH_DTLS=1 MAKE_COAP_DTLS_KEYSTORE=MAKE_COAP_DTLS_KEYSTORE_SIMPLE`

And adding to `project-conf.h` the following line:
```
#define LOG_CONF_LEVEL_COAP		LOG_LEVEL_DBG
#define COAP_DTLS_PSK_DEFAULT_IDENTITY "Client_identity"
#define COAP_DTLS_PSK_DEFAULT_KEY      "secretPSK"
```

But the device does't answer at all. As client I'm testing using `coap-client` from libcoap, and   `Californium`. One strange thing that is happening is that when I use CoAP without DTLS (`coap` instead of `coaps`) the device responds correctly, but it shouldn't as I don't provide the PSK (and not using DTLS). 

To help find out the problem, I'm adding some more informations.

The `viewconf`:
```
Building with tinyDTLS - see ../../../os/net/security/tinydtls/LICENSE
----------------- Make variables: --------------
##### "TARGET": ________________________________ simplelink
##### "BOARD": _________________________________ launchpad/cc1352p1
##### "MAKE_MAC": ______________________________ MAKE_MAC_CSMA
##### "MAKE_NET": ______________________________ MAKE_NET_IPV6
##### "MAKE_ROUTING": __________________________ MAKE_ROUTING_RPL_CLASSIC
##### "MAKE_COAP_DTLS_KEYSTORE": _______________ MAKE_COAP_DTLS_KEYSTORE_SIMPLE
----------------- C variables: -----------------
##### "PROJECT_CONF_PATH": _____________________ == "project-conf.h"
##### "CONTIKI_VERSION_STRING": ________________ == "Contiki-NG-release/v4.2-177-g3642dee5d-dirty"
##### "FRAME802154_CONF_VERSION":_______________ == (0x01)
##### "IEEE802154_CONF_PANID":__________________ == 0xabcd
##### "IEEE802154_CONF_DEFAULT_CHANNEL": _______ == 26
##### "QUEUEBUF_CONF_NUM": _____________________ == 8
##### "NBR_TABLE_CONF_MAX_NEIGHBORS": __________ == 16
##### "NETSTACK_MAX_ROUTE_ENTRIES": ____________ == 16
##### "UIP_CONF_BUFFER_SIZE": __________________ == 1280
##### "UIP_CONF_UDP": __________________________ == 1
##### "UIP_CONF_UDP_CONNS": ____________________ == 8
##### "UIP_CONF_TCP": __________________________ == 0
##### "UIP_CONF_TCP_CONNS": ____________________ == 0
##### "UIP_CONF_ND6_SEND_RA": __________________ == 0
##### "UIP_CONF_ND6_SEND_NS": __________________ == 0
##### "UIP_CONF_ND6_SEND_NA": __________________ == 1
##### "UIP_CONF_ND6_AUTOFILL_NBR_CACHE": _______ == 1
##### "SICSLOWPAN_CONF_FRAG": __________________ == 1
##### "SICSLOWPAN_CONF_COMPRESSION": ___________ == SICSLOWPAN_COMPRESSION_IPHC
##### "ENERGEST_CONF_ON": ______________________ == 0
##### "LOG_CONF_LEVEL_RPL": ____________________ == 0
##### "LOG_CONF_LEVEL_TCPIP": __________________ == 0
##### "LOG_CONF_LEVEL_IPV6": ___________________ == 0
##### "LOG_CONF_LEVEL_6LOWPAN": ________________ == 0
##### "LOG_CONF_LEVEL_NULLNET": ________________ == 0
##### "LOG_CONF_LEVEL_MAC": ____________________ == 0
##### "LOG_CONF_LEVEL_FRAMER": _________________ == 0
##### "LOG_CONF_LEVEL_6TOP": ___________________ == 0
##### "LOG_CONF_LEVEL_COAP": ___________________ == 4
##### "LOG_CONF_LEVEL_LWM2M": __________________ == 0
##### "LOG_CONF_LEVEL_MAIN": ___________________ == 3
------------------------------------------------
```

And attached the uart output from the device, and the wireshark session:
[dtls_log.txt](https://github.com/contiki-ng/contiki-ng/files/2920903/dtls_log.txt)
[dtls_log_wireshark.zip](https://github.com/contiki-ng/contiki-ng/files/2920905/dtls_log_wireshark.zip)

I couldn't make the [LWM2M with DLTS](https://github.com/contiki-ng/contiki-ng/wiki/Tutorial:-LWM2M-and-IPSO-Objects#lwm2m-with-dtls-pre-shared-keys) example work as well. As it uses the CoAP with DTLS as base (without DTLS worked perfectly).

Thanks
Rafael

RCfesk, 2019-03-07:
I had the same problem. In my case, a DTLS_PSK_HINT was passed to the switch/case in **get_psk_info** in the file **coap-uip.c** . This triggered the DTLS error in the bottom of the function. 

I had a look at some old libcoap code, and noticed another case is in that code. Implementing this got the handshake to finish.

`   case DTLS_PSK_HINT:
     /* There is no point in sending a psk_identity_hint hence it is
      * set to zero length. */
  return 0;
`

rnascunha, 2019-03-07:
Thanks for the response @RCfesk . I did get some improvement after a adding the case clause as you said, but its not "really" working.

Now I'm getting the response, but just at the first attempt. Even at that first, checking the log you can see some errors (`PEER NOT FOUND` many times). After that, everything back to the same.

Did you get anything like that?

I'm guessing that there is some incompatibility with the versions of tinyDTLS used by Contiki-NG and libcoap. I will keep checking. At least know I have some ideia were to look!

Adding the logs from both attempts if someone want to look (wireshark and uart output):
[dtls_logs.zip](https://github.com/contiki-ng/contiki-ng/files/2942397/dtls_logs.zip)

Obrigado!

SalvaHasan, 2021-09-28:
Has anyone found a fix for this issue yet? I am trying the cc26x0-web-demo example on cc2650 SensorTag and facing the same issue.

@g-oikonomou @RCfesk

lutgaru, 2021-09-30:
hello, the current dtls code of contiki-ng is obsolete, many improvements have not been ported, I hope to make a pull request soon to solve this, for the moment I send you the most immediate solution in this version of my personal [repository.](https://github.com/lutgaru/contiki-ng/commit/755ceeaac628db0cd7924bf20854540e8e687413)
I hope it is useful for you, I also share this modified version of current tinydtls that technically works in contiki but is not fully tested.(https://github.com/lutgaru/tinydtls/tree/contikingmod)

SalvaHasan, 2021-09-30:
> hello, the current dtls code of contiki-ng is obsolete, many improvements have not been ported, I hope to make a pull request soon to solve this, for the moment I send you the most immediate [...]

Hey, I have already added this in coap-uip.c but it made no difference at all
`case DTLS_PSK_HINT: /* There is no point in sending a psk_identity_hint hence it is * set to zero length. */ `
`return 0;`

lutgaru, 2021-09-30:
True, I had forgotten, there is a serious problem in the dtls version of contiki that was fixed in later versions regarding the version of tinydtls in the first message of the handshake, this solution should fix it for the moment but I recommend you try to use the most current version. (https://github.com/obgm/libcoap/issues/160)

SalvaHasan, 2021-09-30:
Just so that I understand correctly, if I keep 
```
case DTLS_PSK_HINT: 
return 0;
```
in coap-uip.c and add your [version](https://github.com/lutgaru/tinydtls/tree/contikingmod) of tinyDTLS in contiki-ng/os/net/security/ then it should work?

SalvaHasan, 2021-10-13:
I cloned your repository/branch [contiki-ng/dtlsupdate](https://github.com/lutgaru/contiki-ng/tree/dtlsupdate) and then cloned your repository/branch [tinydtls/contikingmod](https://github.com/lutgaru/tinydtls/tree/contikingmod) into /contiki-ng/os/net/security/

After that I still got errors, I managed to fix some errors by moving ecc and sha test files into test folder and also edited ecc.c and ecc.h by changing ecc_ecdsa_sign to ecc_ecdsa_sign_hash this fixed some errors but I still get many 'undefined references" errors during build. I'm attaching a file showing the build errors. Please take a look and see if there's anything I can do to fix this. 

[BuildFail.txt](https://github.com/contiki-ng/contiki-ng/files/7335251/BuildFail.txt)

> True, I had forgotten, there is a serious problem in the dtls version of contiki that was fixed in later versions regarding the version of tinydtls in the first message of the handshake, this solution should fix it for the moment but I recommend you try to use the most current version. ([obgm/libcoap#160 (comment)](https://github.com/obgm/libcoap/issues/160#issuecomment-381536186))

lutgaru, 2021-10-31:
I'm sorry, I haven't compiled this code for a long time, what happens here is that some files of the cc2650 architecture have the same name as the files of the new version of tinyDTLS, now I can't tell you which ones they are besides crypto.c so I still use the workaround. Or you could look for them yourself, they have the same name and that causes conflict.

SalvaHasan, 2021-11-01:
Hello, Yes I was able to compile and successfully run it a few weeks ago after tweaking certain functions and updating file names (including crypto.c)