contiki-ng/contiki-ng #870: RPL- Border router - local access

navan93, 2019-03-10:
I have got the following configuration running.
RPL border router running native on a beaglebone black to which CC1350 launchpad is connected and the mesh network is formed with it.
I am able to ping the mesh nodes from the beaglebone black. However when I try to ping the mesh nodes from another laptop which is in the same LAN as beaglebone black, it doesnt work.

g-oikonomou, 2019-04-16:
You need to enable IPv6 forwarding on your BBB and then you need to set your other computer to use the BBB as a next hop on the way to your mesh IPv6 network. The way to achieve this will vary depending on the operating system your other laptop is using.

I'm closing this, feel free to join us on gitter!

closed 2019-04-16 by g-oikonomou

PranavD10, 2019-05-03:
I am trying to do the same, can you provide the link from where I can know how to do that.