contiki-ng/contiki-ng #876: Support larger RAM sizes in stack-check

g-oikonomou, 2019-03-14:
Stack check assumes that the RAM of the device it's running on is limited to 64KB or less. This is true for all devices we currently support, but for some new CM4 devices with RAM sizes of 160KB or more, the stack-check will need modifed.

Suggest changing return values to `uint32_t`, or perhaps even better using a `typedef`. @atiselsts what do you think?

atiselsts, 2019-03-18:
A good point. Perhaps `int32_t`, to be able to return negative values (as errors)? `ptrdiff_t` would be another option, but that one is difficult to `printf` in a platform independent way.

g-oikonomou, 2019-03-19:
signed is a good idea yes. If we were to use `ptrdiff_t` then I guess we can always cast to signed long to print it?

g-oikonomou, 2019-04-19:
#909

closed 2019-04-19 by g-oikonomou