contiki-ng/contiki-ng #877: TSCH timesource loops

AdamShier1, 2019-03-14:
This problem is related to rpl preferred parent loops but for tsch timesources. 

Say you have a configuration such as:

A----B-----D, where node A is the border router, node B sends to node A (ie node A is his preferred parent and his tsch timesource), and node D sends to node A through B (ie node B is his preferred parent and his tsch timesource). 

If A is restarted, B and D will no longer be able to send messages to him, and in the normal scenario would disassociate and rejoin from node A's new eb. If B is restarted soon after A restarts, B **can** join D's tsch network if he receives an eb from D faster than one from A. Now B and D are eachothers timesources, and will keep eachother looping in a tsch network with no border router, forever waiting for a DIO message which will never come. 

So A is left sending eb's but with noone in the network, and B and D are in the 'old' tsch network, with eachother as their timesources, so the keepalives will be sent and received with no perceived problems.

atiselsts, 2019-03-15:
Would be good to see a Cooja test for this. I'm afraid you're right and the loop will never timeout, since
1) there will be KA packets being sent and ACKed even when B and D do not have any RPL parent;
2) `tsch_rpl_callback_parent_switch` will fail to reset the on D timesource when D resets the RPL parent.

atiselsts, 2019-03-15:
The TSCH timesource loop can form even if the root is not reset. This was discussed recently on Gitter. When a node gets no ACK to a keepalive, it may select another node as it's timesource (the `resynchronize` function). In this situation the RPL routing parent and the TSCH timesource similarly become two different nodes.

AdamShier1, 2019-03-19:
This problem is essentially the same as #608, which in many scenarios was fixed by #769 supressing eb's when you leave a network, the problem is that everyone doesn't necessarily leave the network at the same time, allowing people who left first to sync off others still sending eb's because their still thinking their in a RPL network.

One possibility as mentioned is you dont set a timesource until you've picked a preferred parent, so there would need to be some timer that says if you dont have a preferred parent/rpl network in T time (possibly a function of max DIO timer), then disassociate. Ideally, you would do some kind of 'poisoning' before disassociating to reduce the probability that you rejoin that same loop again.

atiselsts, 2019-03-19:
Thanks for the reference. At least that PR makes sure that the timesource
loops don't stay unresolved forever! That's a very good start.

atiselsts, 2020-05-18:
Fixed in #1166

closed 2020-05-18 by atiselsts