contiki-ng/contiki-ng #878: Problem with new simplelink SDK 2.40.00.81(16 Jan 2019) to CC2652R1F Revision (E)

FernandoWRA, 2019-03-14:
Hi,

I am using contiki-ng with LAUNCHXL-CC26X2R1.

The uC is a CC2652RF1 with Revision (E).

I saw in this site (https://e2e.ti.com/support/wireless-connectivity/bluetooth/f/538/t/767769) that Revision E silicon will only be compatible with SDK v2.40.00.xx and later.

I downloaded and installed the new simplelink. After I did the command:
make CORE_SDK=/home/fernando/ti/simplelink_cc13x2_26x2_sdk_2_40_00_81 TARGET=simplelink BOARD=launchpad/cc26x2r1

I had this messages errors:
========================
  MKDIR     build/simplelink/launchpad/cc26x2r1//obj
  CC        /home/fernando/NG-contiki/contiki-ng/arch/cpu/simplelink-cc13xx-cc26xx/./ccfg-conf.c
  CC        ../../arch/platform/simplelink/cc13xx-cc26xx/platform.c
  CC        ../../arch/platform/simplelink/cc13xx-cc26xx/batmon-sensor.c
  CC        ../../arch/platform/simplelink/cc13xx-cc26xx/launchpad/cc26x2r1/CC26X2R1_LAUNCHXL.c
../../arch/platform/simplelink/cc13xx-cc26xx/launchpad/cc26x2r1/CC26X2R1_LAUNCHXL.c:237:9: error: unknown field 'swiPriority' specified in initializer
         .swiPriority       = 0,
         ^
../../arch/platform/simplelink/cc13xx-cc26xx/launchpad/cc26x2r1/CC26X2R1_LAUNCHXL.c:237:30: error: excess elements in struct initializer [-Werror]
         .swiPriority       = 0,
                              ^
../../arch/platform/simplelink/cc13xx-cc26xx/launchpad/cc26x2r1/CC26X2R1_LAUNCHXL.c:237:30: note: (near initialization for 'ecdhCC26X2HWAttrs[0]')
../../arch/platform/simplelink/cc13xx-cc26xx/launchpad/cc26x2r1/CC26X2R1_LAUNCHXL.c:261:9: error: unknown field 'swiPriority' specified in initializer
         .swiPriority       = 0,
         ^
../../arch/platform/simplelink/cc13xx-cc26xx/launchpad/cc26x2r1/CC26X2R1_LAUNCHXL.c:261:30: error: excess elements in struct initializer [-Werror]
         .swiPriority       = 0,
                              ^
../../arch/platform/simplelink/cc13xx-cc26xx/launchpad/cc26x2r1/CC26X2R1_LAUNCHXL.c:261:30: note: (near initialization for 'ecdsaCC26X2HWAttrs[0]')
../../arch/platform/simplelink/cc13xx-cc26xx/launchpad/cc26x2r1/CC26X2R1_LAUNCHXL.c:285:9: error: unknown field 'swiPriority' specified in initializer
         .swiPriority       = 0,
         ^
../../arch/platform/simplelink/cc13xx-cc26xx/launchpad/cc26x2r1/CC26X2R1_LAUNCHXL.c:285:30: error: excess elements in struct initializer [-Werror]
         .swiPriority       = 0,
                              ^
../../arch/platform/simplelink/cc13xx-cc26xx/launchpad/cc26x2r1/CC26X2R1_LAUNCHXL.c:285:30: note: (near initialization for 'ecjpakeCC26X2HWAttrs[0]')
../../arch/platform/simplelink/cc13xx-cc26xx/launchpad/cc26x2r1/CC26X2R1_LAUNCHXL.c:309:9: error: unknown field 'swiPriority' specified in initializer
         .swiPriority       = 0,
         ^
../../arch/platform/simplelink/cc13xx-cc26xx/launchpad/cc26x2r1/CC26X2R1_LAUNCHXL.c:309:30: error: excess elements in struct initializer [-Werror]
         .swiPriority       = 0,
                              ^
../../arch/platform/simplelink/cc13xx-cc26xx/launchpad/cc26x2r1/CC26X2R1_LAUNCHXL.c:309:30: note: (near initialization for 'sha2CC26X2HWAttrs[0]')
../../arch/platform/simplelink/cc13xx-cc26xx/launchpad/cc26x2r1/CC26X2R1_LAUNCHXL.c:333:9: error: unknown field 'swiPriority' specified in initializer
         .swiPriority       = 0,
         ^
../../arch/platform/simplelink/cc13xx-cc26xx/launchpad/cc26x2r1/CC26X2R1_LAUNCHXL.c:333:30: error: excess elements in struct initializer [-Werror]
         .swiPriority       = 0,
                              ^
../../arch/platform/simplelink/cc13xx-cc26xx/launchpad/cc26x2r1/CC26X2R1_LAUNCHXL.c:333:30: note: (near initialization for 'aesccmCC26XXHWAttrs[0]')
../../arch/platform/simplelink/cc13xx-cc26xx/launchpad/cc26x2r1/CC26X2R1_LAUNCHXL.c:357:9: error: unknown field 'swiPriority' specified in initializer
         .swiPriority       = 0,
         ^
../../arch/platform/simplelink/cc13xx-cc26xx/launchpad/cc26x2r1/CC26X2R1_LAUNCHXL.c:357:30: error: excess elements in struct initializer [-Werror]
         .swiPriority       = 0,
                              ^
../../arch/platform/simplelink/cc13xx-cc26xx/launchpad/cc26x2r1/CC26X2R1_LAUNCHXL.c:357:30: note: (near initialization for 'aesecbCC26XXHWAttrs[0]')
cc1: error: unrecognized command line option '-Wno-unused-const-variable' [-Werror]
cc1: all warnings being treated as errors
../../Makefile.include:342: recipe for target 'build/simplelink/launchpad/cc26x2r1//obj/CC26X2R1_LAUNCHXL.o' failed
make: *** [build/simplelink/launchpad/cc26x2r1//obj/CC26X2R1_LAUNCHXL.o] Error 1
===============================

I need help to solve this problems.

Has anyone solved this issue before and could give me a help?

Thanks.

firmwareguru, 2019-03-15:
The problem here could be that the some fields have changed in the Simplelink SDK 2.40 driver structures.  There are board configuration files in the `arch/platform/simplelink/cc13x2-cc26x2/launchpad/` directories that configure various drivers using structures defined in the SDK but implemented under `arch`.  

There are probably not a lot of changed, but you'll have to go through the build errors and either remove or add references to missing or extra fields as required.

FernandoWRA, 2019-03-18:
I removed first errors, but there are others libraries to change.

After I removed all errors.

The device always reboot after some seconds.

I believe that some libraries about the radio are incompatible with new revision.

I keep trying to change some libraries...

Please, give me more suggestions.

Thanks.

closed 2019-04-16 by g-oikonomou