contiki-ng/contiki-ng #881: Native compilation doesn't work with recent GCC

matiya, 2019-03-15:
Related to #862, when I try to compile the coap-example-server with the following conf:
```
gcc versión 8.2.1 20181127 (GCC) 
Linux default 5.0.1-arch1-1-ARCH #1 SMP PREEMPT Sun Mar 10 10:55:53 UTC 2019 x86_64 GNU/Linux
```

I run into an issue caused by:
```
../../../arch/platform/native/./cfs-posix-dir.c:69:3: error: la salida de ‘strncpy’ puede truncarse copiando 32 bytes desde una cadena de longitud 255 [-Werror=stringop-truncation]
   strncpy(e->name, res->d_name, sizeof(e->name));
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
 If I change strncpy to memcpy in the following files:

> arch/cpu/native/net/tun6-net.c
>arch/platform/native/./cfs-posix-dir.c
`
  // strncpy(e->name, res->d_name, sizeof(e->name));
 memcpy(e->name, res->d_name, sizeof(e->name));
`
It works.

I don't really know what I'm doing, I just google the output and followed the solution in this link: https://github.com/xrootd/xrootd/issues/702.

g-oikonomou, 2019-03-15:
Definitely duplicate of #862 so I'll close this one here

closed 2019-03-15 by g-oikonomou

policeman0077, 2020-08-28:
I encountered same problem on my ubuntu 20.04lts installed which has gcc v9.3.0 .
is there a known gcc version could work with current contiki-ng code? so I can try downgrade my gcc and compile the examples.