contiki-ng/contiki-ng #886: Can't compile Hello world example and flash it to  Zolertia REMOTE

pekoto4349, 2019-03-27:
Hello! i just got the zolertia REMOTE platform and want to try and flash some examples. I started with the hello world example but i get a compilation error. My plan was as such: 
make TARGET=zoul
make TARGET=zoul savetarget
make hello-world.upload
and then login to shell. I get this compilation errors from the 1st make:

  CC        hello-world.c
arm-none-eabi-gcc -Wl,--fatal-warnings -mcpu=cortex-m3 -nostartfiles -T obj_zoul/cc2538.ld -Wl,--gc-sections,--sort-section=alignment -Wl,-Map=hello-world-zoul.map,--cref,--no-warn-mismatch -mthumb -mlittle-endian  hello-world.o   -o hello-world
/usr/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: cannot open linker script file obj_zoul/cc2538.ld: No such file or directory
collect2: error: ld returned 1 exit status
<builtin>: recipe for target 'hello-world' failed
make: *** [hello-world] Error 1
rm hello-world.o

Also i tried :+1: 
$ make TARGET=zoul BOARD=firefly PORT=<your-device-usb-port> hello-world.upload
$ make TARGET=zoul BOARD=firefly PORT=<your-device-usb-port> login

as was described in the Tutorial:Hello,World section but on the 1st make i get:
make: *** No rule to make target 'hello-world.upload'.  Stop.

Can anyone help me figure out what am i doing wrong? Thanks in advance!

g-oikonomou, 2019-04-06:
Hello @pekoto4349 
```
make TARGET=zoul
make TARGET=zoul savetarget
make hello-world.upload
```
These look fine. I tried them locally and they worked

`make TARGET=zoul BOARD=firefly PORT= hello-world.upload`

This is is incorrect. You need to specify something after `PORT`, depending on what device number has been assigned to your firefly. So you need something like `PORT=/dev/ttyACM0`, but you will likely have to replace the `ttyACM0` part. If you are not sure, `tools/motelist` should help.

closed 2019-04-06 by g-oikonomou

pekoto4349, 2019-04-06:
Hello once again! i figured out the problem. Not specifiying the PORT was a typing error while i was writting the issue ( my port was on /dev/ttyUSB0). The documentation guide for the example reffers to REMOTE zolertia though on BOARD specification was firefly (it should have been remote-revb or remote-reva). Also i wanted to enable shell so i added the shell MODULE on makefile but apparently i didnt take into account the dependencies on the makefile and placed it on a wrong line of the makefile.Everything works perfect now. 

Thanks alot for replying!