contiki-ng/contiki-ng #887: Trickle Timer Library Behaviour

edrose, 2019-03-30:
I've noticed some undesirable behaviour in how the trickle timer library handles timers that aren't currently running.

If I call `trickle_timer_inconsistency()` or `trickle_timer_reset_event()` on a timer that has been configured but is not currently running, a new interval will be calculated and a `ctimer` callback scheduled. In the case of timers that have never been set using `trickle_timer_set()` the timer expiry will call a function pointed to by an uninitialised function pointer, causing undefined behaviour if that pointer happens to not be `NULL`. If the timer has been set before, then it'll simply restart the timer with the callback pointer last provided in `trickle_timer_set`. In both cases subsequent calls to `trickle_timer_is_running()` will indicate that the timer is running.

From a user perspective, I had timers running with `NULL` callback function pointers running which did nothing because calls to `trickle_timer_inconsistency()` were made before the timer started. Calling `trickle_timer_set()` would have solved my issue, but this was never called in the application since `trickle_timer_is_running()` indicated that it was already running. This resulted in trickle timer callbacks simply never occurring.

@g-oikonomou

g-oikonomou, 2019-03-30:
I'll confirm this as a bug.

When I wrote the library I was assuming that trickle timers would only be manipulated (reset, inconsistency etc) while running. I am not sure what would be the preferred way to handle this, but clearly something we should fix.

g-oikonomou, 2019-03-30:
So I think we have multiple problems here:

* The library should never set the `ctimer` for a suspended trickle. `ctimer_set()` should only ever happen when the protocol explicitly requests a `trickle_timer_set()`. Therefore I think that  `trickle_timer_inconsistency()` should change such that it does absolutely nothing for suspended trickles.
* The library should behave the same for trickles that have never been started before as it does for trickles that have in the past been running but that are now stopped.
* Lastly and importantly, `trickle_timer_is_running()` should always return a consistent result.

> If I call trickle_timer_inconsistency() or trickle_timer_reset_event() on a timer that has been configured but is not currently running, a new interval will be calculated and a ctimer callback scheduled.

This should not happening, we don't want trickles starting themselves! Do you think a simple check inside `trickle_timer_inconsistency()` would fix this?

```C
void
trickle_timer_inconsistency(struct trickle_timer *tt)
{
  if(tt->i_cur == TRICKLE_TIMER_IS_STOPPED) {
    return;
  }
[...]
}
```

> In the case of timers that have never been set using trickle_timer_set() the timer expiry will call a function pointed to by an uninitialised function pointer, causing undefined behaviour if that pointer happens to not be NULL.

Explicitly do a `tt->cb = NULL` inside `trickle_timer_config()`. I am thinking near L368, where the rest of the fields are set to known values. This will be less important (possibly even irrelevant) if we make sure we never start the ctimer of a suspended trickle.

> If the timer has been set before, then it'll simply restart the timer with the callback pointer last provided in trickle_timer_set. In both cases subsequent calls to trickle_timer_is_running() will indicate that the timer is running.

I think that fixing `trickle_timer_inconsistency()` will take us half way there. I also think we need to explicitly set `i_cur` to `TRICKLE_TIMER_IS_STOPPED` inside `trickle_timer_config()`. So I think in the end `trickle_timer_config()` should look something like this:

```C
uint8_t
trickle_timer_config(struct trickle_timer *tt, clock_time_t i_min,
                     uint8_t i_max, uint8_t k)
{
[...]
  tt->i_min = i_min;
  tt->i_max = i_max;
  tt->i_max_abs = i_min << i_max;
  tt->k = k;
  tt->i_cur = TRICKLE_TIMER_IS_STOPPED;
  tt->cb = NULL;
[...]
}
```

Let me know what you think @edrose

edrose, 2019-03-31:
I think we are broadly on the same page.

>This should not happening, we don't want trickles starting themselves! Do you think a simple check inside trickle_timer_inconsistency() would fix this?

That is exactly how I was considering implementing a fix myself. I think we may need to consider calls to `trickle_timer_consistency()` as well. Whilst it has no actual effect on a stopped trickle timer since `c` gets reset to zero when `new_interval()` is called, it could still potentially print messages to the user that suggest it's having an effect when it isn't.

Would it be worth having a return value on both of those functions? It would not only provide better feedback, but may save the user a call to `trickle_timer_is_running()`. At some points MPL requires inconsistency to be called on the timer if it's running, and for it to be initialised if it isn't. Having that return value could turn code that looks like this:

```c
if (trickle_timer_is_running(tt) {
  trickle_timer_inconsistency(tt);
} else {
  trickle_timer_set(tt, callback, NULL);
}
```

Into something more like this:

```c
if (!trickle_timer_inconsistency(tt)) {
  trickle_timer_set(tt, callback, NULL);
}
```

Which in my opinion, is neater.

>Explicitly do a tt->cb = NULL inside trickle_timer_config(). I am thinking near L368, where the rest of the fields are set to known values. This will be less important (possibly even irrelevant) if we make sure we never start the ctimer of a suspended trickle.

I think that's a good idea. If we're considering though that a timer can be configured once, then stopped/started multiple times would it be better to take the callback as an argument to the `_config()` function rather than to `_set()`? It would prevent the duplication of providing a callback every time the timer is started. I appreciate that this would be a breaking change however, and there may be other use cases that I'm not considering where the current version is more appropriate.

>I think that fixing trickle_timer_inconsistency() will take us half way there. I also think we need to explicitly set i_cur to TRICKLE_TIMER_IS_STOPPED inside trickle_timer_config(). So I think in the end trickle_timer_config() should look something like this:

I hadn't actually noticed that, but I completely agree. The state of the timer seems to be completely defined by `i_cur` so it should be set to `STOPPED` when the timer is configured.

g-oikonomou, 2019-03-31:
> I think we may need to consider calls to trickle_timer_consistency() as well. 

I was thinking about that one too. Yes, let's not increment the consistency counter for suspended trickles.

> Would it be worth having a return value on both of those functions?

Instead of changing the API, how about creating a wrapper function? I was thinking something along the lines of:
```C
trickle_timer_inconsistency_or_set(struct trickle_timer *tt, trickle_timer_cb_t proto_cb, void *ptr)
```

> would it be better to take the callback as an argument to the _config() function rather than to _set()? 

Are we confident that the callback would not change between different "running" cycles? Do we always know what the callback will be at the time of configuring? If the answer is confidently yes to both then we can consider the trade-off between optimizing vs changing the API (not too keen about API changes, unless there is a strong reason).

g-oikonomou, 2019-04-10:
#896

closed 2019-04-10 by g-oikonomou