contiki-ng/contiki-ng #888: Build failure for 1350 sensortag

jsolderitsch, 2019-03-30:
I am following this tutorial: 
[https://sunmaysky.blogspot.com/2018/04/build-and-run-6lbrcontiki-ng-cc26xx-web.html](https://sunmaysky.blogspot.com/2018/04/build-and-run-6lbrcontiki-ng-cc26xx-web.html)
and trying to build the cc26x0-web-demo for the sensortag version. 

I get a build error with the command: 
`make TARGET=cc26x0-cc13x0 BOARD=sensortag/cc1350 MAKE_ROUTING=MAKE_ROUTING_RPL_CLASSIC. `

Error is encapsulated here: 
```
LD build/cc26x0-cc13x0/sensortag/cc1350//cc26x0-web-demo.elf
/usr/local/bin/../lib/gcc/arm-none-eabi/5.2.1/../../../../arm-none-eabi/bin/ld: build/cc26x0-cc13x0/sensortag/cc1350//cc26x0-web-demo.elf section ._user_heap_stack' will not fit in regionSRAM'
/usr/local/bin/../lib/gcc/arm-none-eabi/5.2.1/../../../../arm-none-eabi/bin/ld: region `SRAM' overflowed by 124 bytes
collect2: error: ld returned 1 exit status 
```
When I build the same web demo for the launchpad version of the 1350 I do not get a build error.

Is this expected behavior? Do I need to make some source or makefile adjustments?

g-oikonomou, 2019-03-30:
The resulting firmware image has a very high RAM footprint. You will need to figure out a way to shrink it. Disabling some of the web demo's modules could be a starting point.

Have a look at the example's `project-conf.h`. You can also try reducing the size of some network-related data structures. Check lines 73-75

jsolderitsch, 2019-03-30:
So perhaps, changing the define for `COAP_SERVER` to 0? Line 46 in `project-conf.h`?

Sorry for raising as an issue. I can close it if that is best.

g-oikonomou, 2019-03-30:
The example works fine for this sensortag under the default configuration with RPL Lite. Changing to RPL Classic as per this blog adds overheads. Something has to give. I would start by getting rid of the NET Uart and then if more is needed decide whether you would rather keep MQTT or CoAP.

jsolderitsch, 2019-03-30:
My goal in this is to get some experience with the 3 components of a RDP boarder router, a SLIP radio component, and a sensor node component. Eventually want to be able to use something like IBM Bluemix or ubidots to receive data from the sensortag through the configuration. sunmaysky's blogs were inspirational but always lead to broken or at least confusing behavior after awhile. I am OK with using the default RPL Lite setting and getting 6lbr out of the configuration means that the reason for using Classic goes away?

g-oikonomou, 2019-03-30:
If you do not need 6lbr then you have a choice between RPL Lite or Classic. I'd suggest following the guides in the Contiki-NG wiki and see if you can get a setup that meets your requirements.

jsolderitsch, 2019-03-30:
Will take more time with the Wiki advice and set the blog advice aside for now.

Closing the issue.

closed 2019-03-30 by jsolderitsch