contiki-ng/contiki-ng #889: Orchestra improvement proposal

atiselsts, 2019-03-31:
The way I see it, currently Orchestra suffers from two problems (please correct if my understanding is not right):
1) With the non-storing mode rule (`unicast_per_neighbor_rpl_ns`) the scheduler wakes up the system on every slot. This may severely affect the system's energy consumption or even cause it to avoid deep LPM altogether.
2) With the storing mode rule (`unicast_per_neighbor_rpl_storing`) dedicated links are added only to the timesource node and nodes to which routes are present. This is more efficient. However, RPL does not add routes to peer nodes. These peer nodes also happen to be good candidates for alternative parents and are used as RPL probing targets. As there are no dedicated slots to these nodes, the probing packets are sent on shared slots and suffer from collisions. This may cause RPL parent probing to produce invalid results, long-term suboptimal routing and/or failure to switch to an alternative parent when the link to the main parent becomes bad.

It may be better to add links to all RPL candidate parents. (We should also add functionality for timing out inactive RPL parents.) This way only relevant links will be activated, and there will be less collisions in the setup phase.

This is relevant to the Orchestra receiver-based mode only.

simonduq, 2019-04-04:
I agree with your understanding of Orchestra's current main limitations (next to these is the fact we have just one choffset per slotframe, unlike MSF).

Your suggested fix would be an improvement but what would you think about taking this one step further, with fully dynamic allocation, not limited to any set (e.g. candidate parents) of neighbors? The idea is to add link whenever queueing a packet to a given neighbor. Only thing missing to make this happen is the ability to set the slot operation rtimer between slots, while still accounting for drift compensation.

Mano2Code, 2019-11-29:
Hi , 
Is there any workaround for this .. (With the non-storing mode rule (unicast_per_neighbor_rpl_ns) the scheduler wakes up the system on every slot)

Mano2Code, 2019-11-29:
> The way I see it, currently Orchestra suffers from two problems (please correct if my understanding is not right):
>
> 1. With the non-storing mode rule (`unicast_per_neighbor_rpl_ns`) the scheduler [...]

Hi , 
Is there any workaround for this .. (With the non-storing mode rule (unicast_per_neighbor_rpl_ns) the scheduler wakes up the system on every slot)

atiselsts, 2019-12-17:
Let's continue the discussion here with a broader scope - Orchestra improvements in general.

This is my understanding of the things that would be nice to have for Contiki-NG Orchestra. Some of them have been implemented already in forks of the code.

Orchestra:

- [ ] Multiple channel offsets
- [ ] A rule for link-based scheduling
- [ ] A special rule for the root node(s)
- [ ] Improve the energy consumption of the NS rule in some way (e.g. through adding Tx links dynamically, or through adding Tx links only to neighbors)

TSCH scheduling in general:

- [ ] Adding Tx links dynamically
- [ ] Allow multiple links with the same time offset in the same slotframe
- [ ] Prioritize links based on the the number of packets in their neighbor's queues
- [ ] Time-dependent scheduling
- [ ] More options for slotframe prioritization (e.g. allow to prioritize Rx over Tx)

alexrayne, 2020-04-22:
what is "Time-dependent scheduling" about?

atiselsts, 2020-04-22:
@alexrayne Normally, the same slotframe repeats over time again and again. If there is a conflict / collision between cells to nodes A and B, the conflict stays in the schedule forever.

The idea of time dependent scheduling is to make the slotframe dynamic, so that it does not repeat in the same form.  Each subsequent repetition has the same cells, but arranged in a different way.
It also can be implemented in a secure way, so that the schedule cannot be learned by an attacker, which helps against jamming.

alexrayne, 2020-04-22:
Is it about, you talk before, include TSCH asn into hash?

atiselsts, 2020-04-22:
Yes, that's the current idea how to implement it.

alexrayne, 2020-04-22:
imho, it also can done some work about avoiding slot-frames, that definitely ocupied by other rules. 

For example i see now in my simlation such conflicts with link-based rule:
1) looks like default rule provides one common slot for send broadcasts or to neibohours that yet in discovering state.  smae timeslot used by all neibohours for this
2) this time-slot used by unicast to send ordinary unicast pacet.
3) got situation that in one slot:
```
   mote5   - [uc] ch20-> mote6
   mote6   - [DIO] ch20 -> mote7   <- look this use default rule to send DIO ?
   mote7    -[uc ] ch15 -> mote13
```
here mote7 cant`t get [DIO] on 1st attempt, and receives only second retransmition. therefore this link rating worse, ant it can`t rerouting on shorter path to root. 
More - mote7, can`t receive on 1`st attempt also, since it jumed by mote5.

if mote5 do not jum default slot, and mote7 not use it for own transmition, both - moter5 and 6, have chance to deliver their packet on 1st attempt.

same situation comes with EBs.

alexrayne, 2020-04-22:
imho, default rule may be some more clever/hacky. now all nodes use single time-slot-chanel, and  broadcasts on many nodes jumming each other.

atiselsts, 2020-04-22:
My view is that the default rule is overused. In receiver based mode, unicast packets should not be sent using the default rule / shared slot, they should be sent in the unicast cells. (In other modes, sender and link based, this is more tricky to implement since the receiver node needs to be aware of the sender so that is can schedule a cell for it)