contiki-ng/contiki-ng #890: TSCH timesource selection issues

atiselsts, 2019-03-31:
This is relevant to the #877. Infinite loops may show up in the TSCH timesource tree. Instead of completely hijacking Adam's issue with multiple related problems, here is a new one.

First, a minor comment: the condition in the https://github.com/contiki-ng/contiki-ng/blob/develop/os/net/mac/tsch/tsch.c#L367 seems wrong? It will update the `last_eb` with the current Eb's source address if and only if there is no timesource, or the current `last_eb` address is not the same as the timesource address. As you can see, this will very soon converge to the `last_eb` address being equal to the timesource's address. I'm either misunderstanding something or guessing the if statement is meant to check the frame's source address instead of the `last_eb` address.

Fortunately, the current behavior minimizes the chance that an alternative timesource is selected and so reduces the probability the loop shows up.

Setup to replicate the timesource loop:
Network topology: 1 is the coordinator, 2 and 3 are peer nodes are in direct reachability of 1.
Routing is RPL lite, default settings (`examples/6tisch/simple-node/project-conf.h`).

With the `develop` branch:
* 0:00 - start all nodes
* around 1:00: remove root
* 1:15-1:30 - the node 2 switches to 3 as its timesource and 3 switches to 2 as its timesource
* 1:50 - RPL switches parent to NULL

With the `develop` branch, and after changing the line 367 in `tsch.c`:

    if(ts == NULL || !linkaddr_cmp(&last_eb_nbr_addr, &ts->addr)) {

to this:

    if(ts == NULL || !linkaddr_cmp((linkaddr_t *)&frame.src_addr, &ts->addr)) {

* 0:00 - start all nodes
* around 1:00: remove root
* 1:06 - the node 2 switches to 3 as its timesource and 3 switches to 2 as its timesource
* 12:04 - RPL switches parent from 1 to the peer node (2 or 3)
* 12:04-12:33 - RPL continues switching between no parent (NULL) to the peer node as a parent
* 12:33 - RPL switches parent to NULL

In both cases the TSCH timesource never becomes NULL. This will cause "islands" to form in the network and drift away from the coordinator's clock. Since they never leave TSCH network, they will never get to resync again (unless accidentally drifting back in the TSCH timing and hearing some packets).

My recommendation is to remove the alternative TSCH timesource selection, and in general keep the tsch timesources consistent with RPL parent (e.g. leave the TSCH network when RPL parent becomes NULL).

atiselsts, 2019-03-31:
Looking at the previous discussions, the last bit has already been proposed before: https://github.com/contiki-ng/contiki-ng/issues/608#issuecomment-445093218

debug-ito, 2019-04-02:
if(ts == NULL || !linkaddr_cmp((linkaddr_t *)&frame.src_addr, &ts->addr)) {

I think your proposed condition above is the right one.

simonduq, 2019-04-04:
Agree with fixing the 'if' statement, seems like a bug.

I agree having TSCH switch regardless of RPL comes with downsides, but I've seen it really help in several instances. Makes TSCH take any chance to resync, waiting for RPL to recover.

If RPL does not recover, TSCH should count to infinity and give up when the JP reaches TSCH_MAX_JOIN_PRIORITY. Did that not happen for some reach? Or does that just take too long?

atiselsts, 2019-04-04:
I can't say that I fully understand the role of the join priority, but in this case it does not seem to be relevant. Join priority is propagated in EB messages. When a node leaves the RPL network, it stops sending EB messages (since #769 ). All timesync happens via ACKs, which don't include join priority.

simonduq, 2019-04-04:
Right, this won't work now that we stop sending EBs when RPL leaves. Not sure what the neatest solution is. Feels like when we stop sending EBs we should also leave the TSCH network. Because else we will try to keep syncing with no RPL rank nor TSCH JP, i.e. no protection against loops.

PR #1166 (2020-03-06): keep TSCH timesource state consistent with RPL parent state
Fixes #890 and #877

Potentially fixes #608
closed 2020-03-06 by simonduq