contiki-ng/contiki-ng #894: Acceptable rank calculation

g-oikonomou, 2019-04-06:
In [RPL Classic rpl-dag.c#L1405](https://github.com/contiki-ng/contiki-ng/blob/develop/os/net/routing/rpl-classic/rpl-dag.c#L1405) we make a check to see if our rank via a specific parent would be acceptable.

```C
  if(!acceptable_rank(p->dag, p->rank)) {
    /* The candidate parent is no longer valid: the rank increase resulting
       from the choice of it as a parent would be too high. */
    LOG_WARN("Unacceptable rank %u (Current min %u, MaxRankInc %u)\n", (unsigned)p->rank,
        p->dag->min_rank, p->dag->instance->max_rankinc);
    rpl_nullify_parent(p);
    [...]
  }
```
It strikes me as though we are comparing the candidate parent's rank instead of "ours via them". I would have thought the check should be:

```C
if(!acceptable_rank(p->dag, rpl_rank_via_parent(p))) {
  [...]
}
```

The current logic causes problems when a previously known neighbour suddenly disappears for whatever reason. Its rank increases of course, but our rank via them increases much rapidly due to failed probing.

Am I missing something?

g-oikonomou, 2019-04-06:
> Its rank increases of course, but our rank via them increases much more rapidly due to failed probing.

Actually if they disappear their rank is not increasing at all, we just remember the last known one.

simonduq, 2019-04-06:
I agree and it looks like there's the same mistake in Lite!

closed 2019-04-26 by simonduq