contiki-ng/contiki-ng #897: Multicast CI Tests not testing all Drivers

edrose, 2019-04-12:
I've discovered a bug in the Travis CI tests for the multicast drivers. The test isn't being run for all of the intended. The issue is with how the build system is dealing with the parameters passed to it. The tests I'm referring to are numbers 15-18 in the `tests/07-simulation-base` folder, however it's possible other tests are also affected by this.

The problem stems from how the multicast engines are swapped out between the tests. Setting the multicast engine in contiki-ng is done using the preprocessor macro `UIP_MCAST6_CONF_ENGINE`; this can either be done in `project-conf.h` or using a command line macro definition when building. When cooja builds motes for the tests, the engine is changed using the command line method. When using a clean working directory, this works as expected and the specified engine is used. Cooja creates the mote, and removes the object file for the top-level program (`root.c`/`forwarder.c`/`sink.c`). When the next test is run, make only recompiles the top level program, since that object file was deleted and no other files have changed. This is then linked to the rest of the OS which was compiled for the previous test with the previous test's multicast driver. As a result, the only test being performed is for `roll-tm`, which just happens to be the first test the automated script gets to.

[I've created a working example for this in the `trash/travis-issue` branch in my fork](https://github.com/contiki-ng/contiki-ng/compare/develop...edrose:trash/travis-issue). In this branch, I've deliberately sabotaged the smrf driver so that it never accepts any packets, and this behaviour can be confirmed by running the cooja test at `tests/07-simulation-base/16-cooja-multicast-11-hops-smrf.csc` which will fail when built with a clean build tree. However, when `distclean`ing the `examples/multicast`folder and then running `make` inside of the `tests/07-simulation-base` folder, all tests will pass. [This is reflected in the travis tests that have passed here.](https://travis-ci.org/edrose/contiki-ng/builds/519201020).

The solution to this is to add the `-B` option to the command for creating cooja motes in affected tests, [as I've done in this branch fixing the issue for my MPL implementation.](https://github.com/edrose/contiki-ng/compare/feat/mpl...edrose:bug/travis-tests) I haven't merged this into my feature branch since it's an issue that needs to be fixed separately. There also may be more places where this needs to be fixed that I'm unaware of so I thought I'd create this issue explaining the problem so it can be properly fixed everywhere that it occurs.

edrose, 2019-04-12:
Just had a look and tests 19/10/21 in the `07-simulation-base` folder solve a similar issue to this one by adding a `make clean` command on a newline in the cooja config, like so.

```
<commands EXPORT="discard">make TARGET=cooja clean
 make -j node.cooja TARGET=cooja MAKE_WITH_ORCHESTRA=0 MAKE_WITH_SECURITY=0 MAKE_WITH_PERIODIC_ROUTES_PRINT=1</commands>
```
This has the same effect as using a `-B` flag so either method would work as a solution.

g-oikonomou, 2019-04-12:
I was wondering myself about whether we clean before building. I think we should. Is that something we'd need to add to every individual `.csc` file? Perhaps we should consider forcing it globally somehow?

edrose, 2019-04-12:
I'm not familiar enough with cooja to know whether it can be done globally.

However I don't think it's something that *needs* to be applied to every test, and doing so may just slow the testing process down. It only needs to be done when two tests are being run on the same project folder which have macros that change between the tests, like the multicast tests which are all built in the `examples/multicast` project folder and just change the `UIP_MCAST6_CONF_ENGINE` macro between runs.

g-oikonomou, 2019-04-16:
Hi Ed @edrose 

I just discussed this with the team. We think the best way forward is to globally force `-b` for all cooja tests, but only when we are running inside CI. This would then allow the user to clean/not clean when running a test manually. Niclas @nfi correct me if I am wrong, you mentioned that there was a way to pass things from the environment to Cooja that might help achieve this?

edrose, 2019-04-17:
Thanks for the reply @g-oikonomou.

I've had a flick through the rest of the tests, and the issue about clean building has actually been solved without using the `-B`flag in a lot of other places, and I should have noticed this when initially creating the issue. For example, every test in the `09-ipv6` folder is clean built using something similar to below as the mode build command.

```xml
<commands>make clean
make WITH_CSMA=1 node.cooja</commands>
```
Making the multicast examples fit this format may be an easier way of solving the issue. This technique is used in:

- `07-simulation-base` 03-06 & 19-21
- `09-ipv6/*`
- `13-ieee-802154/*`
- `13-rpl-lite` all except `02-rpl-root-reboot-2.csc`, which doesn't need it
- `14-rpl-classic` same as above
- `17-tun-rpl-br/*`

Having been through all of the other tests, I don't believe there are any other occurrences of the bug I've described  when the tests are run on travis, since each test folder is run independently in a clean tree which minimises the possibility of dirty build directories causing problems.

However, running the tests outside of travis will cause a few more occurrences of the bug to occur. I'm running these by entering the `tests` directory and running `make` inside the contiker container. This doesn't use separate build directories for each test set, and therefore causes problems in:
- `08-native-runs/02-mqtt-client.sh` since it runs on the dirty `examples/mqtt-client` directory used by the `01-compile-base` tests, which had different `DEFINES=` flags.
- `16-rpl-classic/02-rpl-root-reboot-2.csc` since it runs on the dirty `examples/libs/shell` directory that `15-rpl-lite` used in the previous set. This therefore tests RPL lite twice, rather than testing RPL Classic the second time around.

Some of the tests in `17-tun-rpl-br` fail outside of travis due to a seemingly unrelated issue.

g-oikonomou, 2019-04-19:
Thanks

In that case, I suggest changing the mcast tests to clean, as per other similar occurrences. Did you want to open a pull for the mcast tests?

We then need to sit and fix docker...

edrose, 2019-04-20:
I've created pull request #917 to fix the multicast tests. That fixes all of the test issues present within Travis.

Having done some more digging on the issues with local testing, I believe I've sorted them and local testing should act in the same way as Travis testing with a few changes I made in 3db6c964bb90d75c38e88df6522c319cec1e0bcb. The tests I believed were unrelated we actually the same problem, just presenting in a different way.

g-oikonomou, 2019-04-20:
Did you want to open another pull with that commit to fix the local runs?

edrose, 2019-04-21:
I'm just going to run a few more checks. I've worked on the assumption that all of the tests are run in order, but that doesn't seem to be the case. I _think_ it still fixes the issues, however I'm going to do a bit more work on it to be sure

simonduq, 2019-04-23:
So shall we close this issue now?

edrose, 2019-04-23:
I did have one further suggesting for local testing but never haven't had time to investigate it. 

For out of tree testing within the docker image, this only works in an external volume is mounted into `$(HOME)/out-of-tree-tests` with the correct directory structure. Would including that directory structure within the docker image itself be beneficial? It shouldn't break travis tests since external volumes in docker overwrite files in the image IIRC, and it would allow out of tree testing to occur within the docker image with no external volumes.

Then again, out of tree tests may not be something that need to be run locally, and can be done anyway with a little extra work.

Happy to open a separate issue for this if it's better suited elsewhere.

g-oikonomou, 2019-04-26:
So for out-of-tree tests we use https://github.com/contiki-ng/out-of-tree-tests

Travis clones this automatically and mounts it to `$HOME/out-of-tree-tests`. You are of course right that this won't pass when we run tests locally.

I'm not sure at the moment how to suggest handling this

closed 2019-05-10 by g-oikonomou