contiki-ng/contiki-ng #898: Mosquitto broker disconnecting mqtt client

SuFizz, 2019-04-15:
When I try to publish with:

mosquitto_pub -h localhost -t "iot-2/evt/status/fmt/json" -m "1" -i d:secretpassword:mqtt-client:00124bc854ad -u mqtt-client-username -d

My mqtt-client disconnects. And I see that the mosquitto_pub does send a disconnect in the end as well. How do I publish without disconnecting and getting the mqtt-client to get the message "1"?

g-oikonomou, 2019-04-15:
Hi @SuFizz I am not 100% clear what you are trying to describe here. Which mqtt client is it that disconnects, the mosquitto_pub, or the Contiki-NG one? It's normal for mosquitto_pub to disconnect.

On the -NG side, are you using the correct ORG ID as per the readme?

SuFizz, 2019-04-15:
@g-oikonomou : I'm using org_id : secretpassword (some random string). Using quickstart didn't even subscribe (as expected).
Yes, it is the mosquito_pub that sends the disconnect.
I'm following the readme but am having problems with getting the data back to the mqtt-client node. When I send the above mosquito_pub command, it disconnects the mqtt-client with reason 1 (looking at the code that enum just said - it disconnects)
I'm using the contiki-ng default border router (on a raspberry pi)  and slip radio (CC2538) and the nodes are also CC2538.

g-oikonomou, 2019-04-15:
Can you try without -i and -u in your _pub command line and see what happens? It's expected that the mosquitto_pub will disconnect, but I expect the Contiki-NG mqtt-client to stay connected

SuFizz, 2019-04-16:
Nothing happens. The mqtt-client receives nothing. Is the only way to verify that something is being sent through wireshark /  sensniff?

g-oikonomou, 2019-04-16:
I suggest you run the `mosquitto` server with `-v`. You should see the `mqtt-client` connect, subscribe and then start publishing periodically. Once you have verified all that, then I would expect the example to work as advertised.

When your `mosquitto_pub` publishes something, the server should log an attempt to send it to the `mqtt-client`. Feel free to paste the server's log here if you want.

SuFizz, 2019-04-16:
The log is here:

1555088711: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1555088711: Using default config.
1555088711: Opening ipv4 listen socket on port 1883.
1555088711: Opening ipv6 listen socket on port 1883.
1555088715: New connection from fd00::212:4b00:14c8:54ad on port 1883.
1555088723: New client connected from fd00::212:4b00:14c8:54ad as d:secretpassword:mqtt-client:00124bc854ad (c1, k11520, u'mqtt-client-username').
1555088723: Sending CONNACK to d:secretpassword:mqtt-client:00124bc854ad (0, 0)
1555088723: Received SUBSCRIBE from d:secretpassword:mqtt-client:00124bc854ad
1555088723: 	iot-2/cmd/+/fmt/json (QoS 0)
1555088723: d:secretpassword:mqtt-client:00124bc854ad 0 iot-2/cmd/+/fmt/json
1555088723: Sending SUBACK to d:secretpassword:mqtt-client:00124bc854ad
1555088724: New connection from ::1 on port 1883.
1555088724: New client connected from ::1 as mosqsub/990-raspberrypi (c1, k60).
1555088724: Sending CONNACK to mosqsub/990-raspberrypi (0, 0)
1555088724: Received SUBSCRIBE from mosqsub/990-raspberrypi
1555088724: 	# (QoS 0)
1555088724: mosqsub/990-raspberrypi 0 #
1555088724: Sending SUBACK to mosqsub/990-raspberrypi
1555088763: Received PUBLISH from d:secretpassword:mqtt-client:00124bc854ad (d0, q0, r0, m0, 'iot-2/evt/status/fmt/json', ... (224 bytes))
1555088763: Sending PUBLISH to mosqsub/990-raspberrypi (d0, q0, r0, m0, 'iot-2/evt/status/fmt/json', ... (224 bytes))
1555088784: Received PINGREQ from mosqsub/990-raspberrypi
1555088784: Sending PINGRESP to mosqsub/990-raspberrypi
1555088786: Received PUBLISH from d:secretpassword:mqtt-client:00124bc854ad (d0, q0, r0, m0, 'iot-2/evt/status/fmt/json', ... (224 bytes))
1555088786: Sending PUBLISH to mosqsub/990-raspberrypi (d0, q0, r0, m0, 'iot-2/evt/status/fmt/json', ... (224 bytes))
**_1555088808_**: New connection from ::1 on port 1883.
1555088808: New client connected from ::1 as mosqpub/1370-raspberryp (c1, k60).
1555088808: Sending CONNACK to mosqpub/1370-raspberryp (0, 0)
1555088808: Received PUBLISH from mosqpub/1370-raspberryp (d0, q0, r0, m0, 'iot-2/evt/status/fmt/json', ... (1 bytes))
1555088808: Sending PUBLISH to mosqsub/990-raspberrypi (d0, q0, r0, m0, 'iot-2/evt/status/fmt/json', ... (1 bytes))
1555088808: Received DISCONNECT from mosqpub/1370-raspberryp
1555088808: Client mosqpub/1370-raspberryp disconnected.
1555088815: Received PUBLISH from d:secretpassword:mqtt-client:00124bc854ad (d0, q0, r0, m0, 'iot-2/evt/status/fmt/json', ... (224 bytes))
1555088815: Sending PUBLISH to mosqsub/990-raspberrypi (d0, q0, r0, m0, 'iot-2/evt/status/fmt/json', ... (224 bytes))

At point : 1555088808 is where I do the mosquitto_pub like you had suggested ie without the -i and -u.

g-oikonomou, 2019-04-16:
I see the problem now. The `mosquitto_pub` publishes to the wrong topic.

It publishes to `iot-2/evt/status/fmt/json`

It should be publishing to `iot-2/cmd/+/fmt/json`

Note the difference: `cmd` vs `evt`. So you basically need to adjust your `-t` argument.

Please test this and close the issue if it works.

SuFizz, 2019-04-16:
Thank you!! Solves the problem!

closed 2019-04-16 by SuFizz