contiki-ng/contiki-ng #901: Simplelink target not sleeping (standby)

RCfesk, 2019-04-16:
I'm not able to get the leaf nodes to sleep using the Simplelink on the CC1312.
What I observe is that the function Power_getConstraintMask() returns 4, meaning that PowerCC26XX_DISALLOW_STANDBY is set for some reason.  I try to pinpoint where this is set, but it seems to be set in a precompiled ISR or something.

I set 
#define RPL_CONF_DEFAULT_LEAF_ONLY              1
to enable only leaf node operation. 

Am I misunderstanding that the node is supposed to auto-sleep in this configuration? 
The code runs as expected else wise, even up to CoAP functionality.

The configuration is as follows:
```
##### "PROJECT_CONF_PATH": _____________________ == "project-conf.h"
##### "CONTIKI_VERSION_STRING": ________________ == "Contiki-NG-release/v4.2-dirty"
##### "FRAME802154_CONF_VERSION":_______________ == (0x01)
##### "IEEE802154_CONF_PANID":__________________ == 0x9812
##### "IEEE802154_CONF_DEFAULT_CHANNEL": _______ == 20
##### "QUEUEBUF_CONF_NUM": _____________________ == 16
##### "NBR_TABLE_CONF_MAX_NEIGHBORS": __________ == 16
##### "NETSTACK_MAX_ROUTE_ENTRIES": ____________ == 1024
##### "UIP_CONF_BUFFER_SIZE": __________________ == 1280
##### "UIP_CONF_UDP": __________________________ == 1
##### "UIP_CONF_UDP_CONNS": ____________________ == 8
##### "UIP_CONF_TCP": __________________________ == 0
##### "UIP_CONF_TCP_CONNS": ____________________ == 0
##### "UIP_CONF_ND6_SEND_RA": __________________ == 0
##### "UIP_CONF_ND6_SEND_NS": __________________ == 0
##### "UIP_CONF_ND6_SEND_NA": __________________ == 1
##### "UIP_CONF_ND6_AUTOFILL_NBR_CACHE": _______ == 1
##### "SICSLOWPAN_CONF_FRAG": __________________ == 1
##### "SICSLOWPAN_CONF_COMPRESSION": ___________ == SICSLOWPAN_COMPRESSION_IPHC
##### "ENERGEST_CONF_ON": ______________________ == 0
##### "LOG_CONF_LEVEL_RPL": ____________________ == 0
##### "LOG_CONF_LEVEL_TCPIP": __________________ == 0
##### "LOG_CONF_LEVEL_IPV6": ___________________ == 0
##### "LOG_CONF_LEVEL_6LOWPAN": ________________ == 0
##### "LOG_CONF_LEVEL_NULLNET": ________________ == 0
##### "LOG_CONF_LEVEL_MAC": ____________________ == 0
##### "LOG_CONF_LEVEL_FRAMER": _________________ == 0
##### "LOG_CONF_LEVEL_6TOP": ___________________ == 0
##### "LOG_CONF_LEVEL_COAP": ___________________ == 0
##### "LOG_CONF_LEVEL_LWM2M": __________________ == 0
##### "LOG_CONF_LEVEL_MAIN": ___________________ == 4
```

```
[INFO: Main      ] Starting Contiki-NG-release/v4.2-dirty
[INFO: Main      ] - Routing: RPL Lite
[INFO: Main      ] - Net: sicslowpan
[INFO: Main      ] - MAC: CSMA
[INFO: Main      ] - 802.15.4 PANID: 0x9812
[INFO: Main      ] - 802.15.4 Default channel: 20
[INFO: Main      ] Node ID: 28220
[INFO: Main      ] Link-layer address: 0012.4b00.18d3.6e3c
[INFO: Main      ] Tentative link-local IPv6 address: fe80::212:4b00:18d3:6e3c
[DBG : CC13xx/CC26xx] With DriverLib v0.51995
[DBG : CC13xx/CC26xx] IEEE 802.15.4: No, Sub-1 GHz: No, BLE: No
[INFO: CC13xx/CC26xx] Operating frequency on Sub-1 GHz
[INFO: CC13xx/CC26xx] RF: Channel 20[INFO: CC13xx/CC26xx] 
[INFO: CC13xx/CC26xx] Node ID: 28220
```

RCfesk, 2019-04-26:
As a related question: Is it possible to run Simplelink in tickless mode or is there a simple way to change the tick frequency without too many side effects?

tiepettersen, 2019-05-02:
Tickless mode implemented here: #943

closed 2019-05-05 by g-oikonomou