contiki-ng/contiki-ng #905: make: *** No rule to make target 'hello-world.bin'.  Stop.

kainataltaf, 2019-04-17:
I am trying to make a bin file of hello world program to run on my MCU. I am writing the following line:
**make TARGET=cc26x0-cc13x0 BOARD=sensortag/cc2650 PORT=/dev/ttyACM0 hello-world.bin**
and it gives me this error:
**make: *** No rule to make target 'hello-world.bin'.  Stop.**
Similar error arise when I try to make a bin file of cc26x0-demo file.
I have updated the sub modules using "git submodule update --init". 
Is there any other way to compile the code and bin file for cc26x0-cc13x0? What am I doing wrong?

closed 2019-04-17 by kainataltaf

sogggy, 2022-01-25:
Hello, I am having the same issue, what did you do to fix this? Thanks!

alexrayne, 2022-01-25:
for arm cpu provided linking of elf image. in my practice - it enough.

this can be easy converted into rec image, and some targets have this.

msp430 have rule for generate ihex, don't know - if it can be ported to any gcc or platform specific.

if you want generate bin from make - patch it!

alexrayne, 2022-01-25:
what happen if you builds:
`make TARGET=cc26x0-cc13x0 BOARD=sensortag/cc2650 PORT=/dev/ttyACM0 hello-world`

??

sogggy, 2022-01-25:
Hi, I managed to solve the issue after updating the submodules. Thanks!