contiki-ng/contiki-ng #910: RPL: DIO may not be sent if there is continuous DIS.

hseokman, 2019-04-18:
I was testing a large number of device simulation and many devices did not join after long period of time. The issue was the root device was not sending DIO.

rpl_timers_dio_reset in os/net/routing/rpl-lite/rpl_timers.c is getting called every time a DIS is received.  This sets a new timer to send DIO. When there are many devices sending DIS, it is enough that the DIO sending timer never expires, so the DIO is not sent.

Is it necessary to create the timer every time is it reset? Wouldn't it be better to leave already running timer if it was started with the minimum DIO interval?

pciolkosz, 2019-07-20:
I've encountered this issue as well. The fix seems to be really simple. Wouldn't it be enough to add something like:
`if (!ctimer_expired(&curr_instance.dag.dio_timer))
  return;
`
at the start of new_dio_interval function?

yatch, 2019-07-23:
@hseokman Could you try PR #1015 to see if the change, https://github.com/contiki-ng/contiki-ng/pull/1015/commits/06e817498958f5717d7a8154e44624a9b3a69abe, resolves the issue?

PR #1015 (2019-09-25): RPL-Lite: don't reset DIO timer if the current interval is equal to Imin
Otherwise, heavy traffic of DIS prevents routers from sending DIOs. This issue is found only in `rpl-lite`. 

Fixes #910
closed 2019-09-25 by simonduq