contiki-ng/contiki-ng #920: Tutorial: TSCH and 6TiSCH - doesn't work

StefanHri, 2019-04-22:
I followed the tutorial https://github.com/contiki-ng/contiki-ng/wiki/Tutorial:-TSCH-and-6TiSCH using two cc2650 devices. 
Unfortunately my not-coordinator device never gets associated. For the not-coordinator device I have the MAC logging enabled. I get the following output:

`...
[INFO: TSCH      ] scanning on channel 15␊
[INFO: TSCH      ] scanning on channel 20␊
[INFO: TSCH      ] scanning on channel 20␊
[INFO: TSCH      ] scanning on channel 20␊
[INFO: TSCH      ] scanning on channel 25␊
[INFO: TSCH      ] scanning on channel 25␊
[INFO: TSCH      ] scanning on channel 15␊
[INFO: TSCH      ] scanning on channel 25␊
[INFO: TSCH      ] scanning on channel 26␊
[INFO: TSCH      ] scanning on channel 25␊
[INFO: TSCH      ] scanning on channel 26␊
[INFO: TSCH      ] scanning on channel 25␊
[INFO: TSCH      ] scanning on channel 15␊
[INFO: TSCH      ] scanning on channel 20␊
[INFO: TSCH      ] scanning on channel 20␊
[INFO: TSCH      ] scanning on channel 15␊
[INFO: TSCH      ] scanning on channel 20␊
[INFO: TSCH      ] scanning on channel 15␊
[INFO: TSCH      ] scanning on channel 15␊
[INFO: TSCH      ] scanning on channel 20␊
[INFO: TSCH      ] scan: received packet (24 bytes) on channel 20␊
[INFO: TSCH Pkt  ] ! parse_eb: frame is not a valid TSCH beacon. Frame version 2, type 1, FCF 41 e9␊
[INFO: TSCH Pkt  ] ! parse_eb: frame was from 0xabcd/0012.4b00.12b9.c702 to 0xabcd/ffff.0000.0000.0000␊
[ERR : TSCH      ] ! failed to parse EB (len 24)␊
[INFO: TSCH      ] scanning on channel 26␊
[INFO: TSCH      ] scanning on channel 20␊
[INFO: TSCH      ] scanning on channel 20␊
...`

I assume that the that the EB  is received because 0012.4b00.12b9.c702 is the address of my coordinator but something else is wrong.

yatch, 2019-04-22:
> [INFO: TSCH Pkt  ] ! parse_eb: frame is not a valid TSCH beacon. Frame version 2, type 1

That received frame is not an enhanced beacon frame since its frame type is `1` (`FRAME802154_DATAFRAME`). Wait for a little bit more to see if the node can receive an EB from your coordinator.

Using a single channel could make this simple test done faster.

StefanHri, 2019-04-22:
Thank you for the fast response Yasuyuki. Using a single channel (20) the result is unfortunatly the same. 
In addition I noticed also one different warning from TSCH:

```c
[INFO: TSCH      ] scanning on channel 20␊
[INFO: TSCH      ] scan: received packet (24 bytes) on channel 20␊
[WARN: TSCH      ] scan: dropping packet, timestamp too far from current time 25342169 25342918␊
```

FelipeGdM, 2019-05-08:
I'm facing the same issue. I have 3 cc2650 launchpad, one of them acting as border router of the RPL mesh. When they boot roughly at the same time, the TSCH sync is done fast and the RPL net is set without major problems, but when one of the motes is restarted it takes about 10 minutes to resync along the others.

Looking the logs, the error is the same from @StefanHri 

```
[INFO: TSCH      ] scanning on channel 25
[INFO: TSCH      ] scan: received packet (62 bytes) on channel 25
[INFO: TSCH Pkt  ] ! parse_eb: frame is not a valid TSCH beacon. Frame version 1, type 1, FCF 41 d8
[INFO: TSCH Pkt  ] ! parse_eb: frame was from 0x1b1/0012.4b00.0f24.c885 to 0x1b1/ffff.0000.0000.0000
[ERR : TSCH      ] ! failed to parse EB (len 62)
[INFO: TSCH      ] scan: received packet (62 bytes) on channel 25
[WARN: TSCH      ] scan: dropping packet, timestamp too far from current time 4134703 4136456
[INFO: TSCH      ] scan: received packet (62 bytes) on channel 25
[WARN: TSCH      ] scan: dropping packet, timestamp too far from current time 4134995 4137412
[INFO: TSCH      ] scan: received packet (62 bytes) on channel 25
[WARN: TSCH      ] scan: dropping packet, timestamp too far from current time 4135287 4138368
[INFO: TSCH      ] scan: received packet (62 bytes) on channel 25
[WARN: TSCH      ] scan: dropping packet, timestamp too far from current time 4135579 4139326
[INFO: TSCH      ] scan: received packet (62 bytes) on channel 25
[WARN: TSCH      ] scan: dropping packet, timestamp too far from current time 4136455 4140282
```

Before that, it had received a EB from a neighbord

```
[INFO: TSCH      ] scanning on channel 25
[INFO: TSCH      ] scan: received packet (35 bytes) on channel 25
[INFO: TSCH      ] parse_eb: no schedule, setting up minimal schedule
[INFO: TSCH Sched] add_slotframe 0 5
[INFO: TSCH Sched] add_link sf=0 opt=Tx|Rx|Sh type=ADV ts=0 ch=0 addr=ffff.ffff.ffff.ffff
[INFO: TSCH Queue] update time source: (NULL LL addr) -> 0012.4b00.0f83.1983
[INFO: TSCH      ] association done (1), sec 0, PAN ID e0e, asn-0.198e3, jp 1, timeslot id 0, hopping id 0, slotframe len 0 with 0 links, from 0012.4b00.0f83.1983
```

Here, ```0012.4b00.0f83.1983``` is the border router. Despite the association, it soon losses sync because it doesn't receives an ACK for the keep alive message

```
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 214, status 0, tx 1
[INFO: Frame 15.4] Out:  1 0012.4b00.0f83.1983 21 0 (21)
[INFO: TSCH      ] send packet to 0012.4b00.0f83.1983 with seqno 44, queue 1 1, len 21 21
[INFO: TSCH      ] sending KA to 0012.4b00.0f83.1983
[INFO: TSCH      ] packet sent to 0012.4b00.0f83.1983, seqno 44, status 2, tx 8
[INFO: TSCH      ] KA sent to 0012.4b00.0f83.1983, st 2-8
[WARN: TSCH      ] not able to re-synchronize, received no EB from other neighbors
[WARN: TSCH      ] leaving the network, stats: tx 9, rx 0, sync 0
[INFO: TSCH Queue] update time source: 0012.4b00.0f83.1983 -> (NULL LL addr)
```

After that, I start receiving the same error ```droping packet```

@StefanHri, in your case, does the non-coordinator mote try to send a keep alive message?

Thank you for the attention

FelipeGdM, 2019-05-08:
The fix described here #657 worked pretty well. 
The resync time dropped from ~10 min to ~1 min

NeoN7777, 2019-05-14:
Hi, I have absolutly the same problem as @StefanHri 

`[INFO: TSCH      ] scanning on channel 25
[INFO: TSCH      ] scanning on channel 25
[INFO: TSCH      ] scanning on channel 26
[INFO: TSCH      ] scanning on channel 26
[INFO: TSCH      ] scanning on channel 26
[INFO: TSCH      ] scan: received packet (24 bytes) on channel 26
[INFO: TSCH Pkt  ] ! parse_eb: frame is not a valid TSCH beacon. Frame version 2, type 1, FCF 41 e9
[INFO: TSCH Pkt  ] ! parse_eb: frame was from 0xabcd/0012.4b00.17e6.b5ff to 0xabcd/ffff.0000.0000.0000
[DBG : TSCH      ] ! failed to parse EB (len 24)
[INFO: TSCH      ] scanning on channel 26
[INFO: TSCH      ] scanning on channel 15
[INFO: TSCH      ] scanning on channel 25
[INFO: TSCH      ] scanning on channel 15
[INFO: TSCH      ] scanning on channel 20
[INFO: TSCH      ] scanning on channel 15
[INFO: TSCH      ] scanning on channel 25
`
0012.4b00.17e6.b5ff - is the valid ip address of the another mote
Any suggestions? Thanks.

FelipeGdM, 2019-05-14:
@NeoN7777 , have you tried decreasing the TSCH_CONF_KEEPALIVE_TIMEOUT ?
```c
#define TSCH_CONF_KEEPALIVE_TIMEOUT CLOCK_SECOND
```

NeoN7777, 2019-05-14:
> @NeoN7777 , have you tried decreasing the TSCH_CONF_KEEPALIVE_TIMEOUT ?
> 
> ```c
> #define TSCH_CONF_KEEPALIVE_TIMEOUT CLOCK_SECOND
> ```
@FelipeGdM Yes, i made it 9 seconds, nothing changed.

aalami90, 2019-06-28:
> > @NeoN7777 , have you tried decreasing the TSCH_CONF_KEEPALIVE_TIMEOUT ?
> > ```c
> > #define TSCH_CONF_KEEPALIVE_TIMEOUT CLOCK_SECOND
> > ```
> 
> @FelipeGdM Yes, i made it 9 seconds, nothing changed.

Did you guys find the solution, I also made it 8 seconds and I still get the same exact errors; I am also using CC2650 Sensortags

NeoN7777, 2019-06-28:
> Did you guys find the solution, I also made it 8 seconds and I still get the same exact errors; I am also using CC2650 Sensortags

No, I'm stuck in this problem, there was another tasks. If you'll find the solution, pls let us know ;)

aalami90, 2019-06-28:
> > Did you guys find the solution, I also made it 8 seconds and I still get the same exact errors; I am also using CC2650 Sensortags
> 
> No, I'm stuck in this problem, there was another tasks. If you'll find the solution, pls let us know ;)

Ya definitely, Thanks for your answer :)

aalami90, 2019-07-01:
> > Did you guys find the solution, I also made it 8 seconds and I still get the same exact errors; I am also using CC2650 Sensortags
> 
> No, I'm stuck in this problem, there was another tasks. If you'll find the solution, pls let us know ;)

Alright problem solved, all I did was make the TSCH coordinator an RPL root; then the motes joined the network automatically.....the problem was that no Enhanced Beacons were being sent by the TSCH coordinator (without an RPL root-node, no "routing" happens...so no beacons were sent)...let me know if it works for you.

closed 2019-08-31 by g-oikonomou

Eden-Contiki, 2020-02-07:
I tried to to implement two nodes to be able to communicate each other and enabling TSCH with  6TiSCH minimal schedule. the mote that I used to test is Zolertia dual motes. I get the following output:

TSCH: scanning on channel 25
TSCH: scanning on channel 26
TSCH: scanning on channel 20
TSCH: scanning on channel 15
TSCH: scanning on channel 25
TSCH: scanning on channel 26
TSCH: scanning on channel 20
TSCH: scanning on channel 15
TSCH: scanning on channel 25
TSCH: scanning on channel 26
TSCH: scanning on channel 20
TSCH: scanning on channel 15
TSCH: scanning on channel 25
TSCH: scanning on channel 26
TSCH: scanning on channel 20
TSCH: association: received packet (8 bytes) on channel 20
TSCH:! parse_eb: frame is not a valid TSCH beacon. Frame version 0, type 3, FCF 03 08
TSCH:! parse_eb: frame was from 0x0/0.0 to 0xffff/255.255
TSCH:! failed to parse EB (len 8)

is  anybody can have any suggestion? Thanks.

atiselsts, 2020-02-10:
@Eden-Contiki in your network, do you have a coordinator device that is sending out EB?

Eden-Contiki, 2020-02-11:
Yes i have

but i have already solve the problem with another setup.

thank you

aaa-engi, 2020-08-15:
I am new in 6tisch 
how i can add an centralized scheduling algorithm to 6tisch?
do you have sample for it?
please help me...

lpn22, 2021-06-17:
i followed tutorial TSCH Contiki NG using 2 cc2538dk, everything is fine but when login it has no response.  Maybe RAM footprint isn't enough, t tried to reduce the memory usage but it still doesn't work. please help me:

user@iot-workshop:~/contiki-ng/examples/hello-world$ make TARGET=cc2538dk PORT=/dev/ttyUSB0 login
rlwrap ../../tools/serial-io/serialdump -b115200 /dev/ttyUSB0
connecting to /dev/ttyUSB0 [OK]