contiki-ng/contiki-ng #933: TSCH is sending EB when the RPL prefered parent is NULL

atiselsts, 2019-04-26:
In RPL classic, the `joined` flag of a dag might be set even if the node does not have a preferred parent. As a result, TSCH may continue to send EB and create infinite timesource loops. I did not investigate whether this problem also applies to RPL lite.

According to [RFC 8180](https://tools.ietf.org/html/rfc8180#section-6.3):

>  When a RPL node joins the network, it MUST NOT send EBs before having
>   acquired a RPL Rank to avoid inconsistencies in the time
>   synchronization structure.  This applies to other routing protocols
>   with their corresponding routing metrics.  As soon as a node acquires
>   routing information (e.g., a RPL Rank, see Section 5.1.1), it SHOULD
>   start sending EBs.

Since the spirit / desired outcome of this paragraph is "to avoid inconsistencies in the time synchronization structure" I think that the RPL "has joined" functionality should be improved to also take into account parent state.

debug-ito, 2019-05-09:
In rpl-lite, the dag object goes to the JOINED state after it finds the preferred parent (https://github.com/contiki-ng/contiki-ng/blob/fb6451ef7b3f32e12d3ec0d7fe1de34476e4b4a0/os/net/routing/rpl-lite/rpl-dag.c#L330 ).

g-oikonomou, 2019-10-26:
Fixed in #1106

PR #1106 (2019-10-26): RPL CLassic: only allow DAGs with a parent in rpl_has_joined
Fixes #933
Fixes #1105
closed 2019-10-26 by g-oikonomou