contiki-ng/contiki-ng #935: Joining a BR using broadcast PAN ID

PranavD10, 2019-04-29:
Hi,

I am working on a project which where the Node would like to be added after the network infrastructure has been created. 
Let me elaborate little more, Let say I have one currently active network with the PAN ID 0xABCD. Now I want to add one new node into the current network. I have configured this node to broadcast to all the network using PAN ID 0xFFFF. Now at this point, BR can get the messages from the new node. But how will I configure new node to change its PAN id to the existing network PAN ID.
So is there any way that the two pan ids 0XABCD and 0XFFFF can communicate with each other and send the messages.
Ping me if u need more details.

Thank you

yatch, 2019-04-29:
You don't need to change the PAN ID to 0xFFFF. The new node which configured 0xABCD as its PAN ID should be receive a (beacon) frame which has 0xFFFF in its destination address field.

I'm closing this issue. Feel free to reopen if it doesn't work.

closed 2019-04-29 by yatch

PranavD10, 2019-04-29:
Thank you for replying,
Is there any provision, I can send the messages from 0xABCD to 0xFFFF.

yatch, 2019-04-29:
I don't think so. I'm not sure why you need it, though. If you want to do such a thing, I suggest see how MAC layer is implemented, which may be interesting to you. For example, this part: https://github.com/contiki-ng/contiki-ng/blob/develop/os/net/mac/framer/framer-802154.c#L196

PranavD10, 2019-04-30:
This pull request is there for broadcasting on the PAN NETWORKS. Its in the development and may be I have look for something else 
https://github.com/contiki-ng/contiki-ng/pull/636
Would be grateful for your suggestion.
Thank you