contiki-ng/contiki-ng #936: Communication between the Broadcast PAN ID and the network configured PAN ID

PranavD10, 2019-04-29:
Hi,
I am working on the project in which nodes can be dynamically added to the network. Also the nodes don't have the information about the PAN ID and all about the existing network. so If I broadcast the info from node using PAN ID 0XFFFF it will be received by the BR working on 0XABCD. now from this BR I intend to send the info on 0XFFFF to the node and tell it to switch to the new PAN ID.
I want to know if it is possible to do that.
I know the broadcasting on networks using 0xffff is still in the development but still if it is possible...???

Thanks

yatch, 2019-04-29:
If you're using TSCH, it seems a joining node can join any PAN ID: https://github.com/contiki-ng/contiki-ng/blob/develop/os/net/mac/tsch/tsch-conf.h#L184

I guess this is the case for CSMA as well since it seems there is no check on received PAN ID.

I'm closing tihs issue. If it didn't work in your experiment, feel free to reopen the issue with the exact configuration you have.

closed 2019-04-29 by yatch

PranavD10, 2019-04-29:
Hi yatch,
I am using CSMA , I have performed the experiment in which I set up two border router with different PAN IDs and I observed that the node only joins the PAN ID for which it is configured.
and my question is will that pan id can send the data to the broadcast PAN ID 0xFFFF

Thank you for replying

yatch, 2019-04-29:
Seems the (radio) hardware filters incoming frames with the configured PAN ID. `grep` with `RADIO_PARAM_PAN_ID` to find relevant code to see, and consult the user document of your hardware if you need.