contiki-ng/contiki-ng #938: Link stats: unsuccessful probing can increase link stats etx metric

atiselsts, 2019-04-30:
Let's say we have topology:
 
    RPL root <-> A <-> B, C, D, E

Sequence of events:
- the RPL root disappears
- node A continuously increases etx to root and advertises it to downstream nodes
- node A leaves the TSCH network and deletes all link stats (let's say it has `etx_1` to root)
- node A rejoins TSCH and RPL networks through some of the downstream nodes
- node A sends a probing message to root
- node A fails to get reply to the probing message
- node A adds a new link stats with some default etx + noack penalty (let's call it `etx_2`)

Now even though the probing to root did not get any reply, `etx_2` is likely to be lower than `etx_1`. This will cause the downstream nodes to decrease their etx to root as well.

Expected behavior: the RPL etx metrics reach higher and higher numbers due to counting to infinity; all nodes eventually leave the RPL and TSCH networks.

Actual behavior: counting to infinity is slow or does not happen, as the etx keeps resetting.

simonduq, 2019-10-18:
Postponing: this needs more discussion before we converge on a solution, for 4.5

debug-ito, 2019-12-24:
I have some questions about the description.

- What do you mean by "node A rejoins TSCH and RPL networks through some of the downstream nodes"? Is it that node A hears EBs and DIOs from node B (for example)?
- When "node A rejoins TSCH and RPL networks", is the root still dead? Or, the root has resetted and is now working?

If the root is still dead, the essential problem is the TSCH synchronization loop (#877). In addition, why does node A send probes to the root? Node A already left TSCH and RPL network once and heard DIOs from node B but not from the root, right? I think there is no entry for the root in node A's neighbor table, so node A doesn't send anything to the root. Maybe I'm missing something about behavior of a node..

If the root is working again, I think it makes sense to set a default value to its ETX on node A. That's because it's working and near node A. However, TSCH synchronization loop makes it tricky. That is, even if the root is working and near node A, it's possible that node A cannot hear the root if node A is trapped in a TSCH synchronization loop with node B.

atiselsts, 2019-12-24:
>  What do you mean by "node A rejoins TSCH and RPL networks through some of the downstream nodes"? Is it that node A hears EBs and DIOs from node B (for example)?

Yes

>  When "node A rejoins TSCH and RPL networks", is the root still dead? Or, the root has resetted and is now working?

Let's assume that the root is dead / removed. Maybe the whole network is moved to a new location, and the nodes should leave the old TSCH network and join the root node at the new location.

> In addition, why does node A send probes to the root? 

The node A still has all old neighbors in the RPL parent table. We were using RPL classic, by the way.

debug-ito, 2019-12-25:
I see. Thanks for explanation.

> Maybe the whole network is moved to a new location

So, the original root is removed and a new root (with different MAC address, IPv6 prefix and address) is introduced.

> The node A still has all old neighbors in the RPL parent table. We were using RPL classic, by the way.

Hmm, I haven't used rpl-classic. In rpl-lite, `rpl_local_repair` and `rpl_dag_leave` functions call `rpl_neighbor_remove_all`. So, if a node leaves an RPL network, its neighbor table becomes empty, I think. Maybe rpl-classic should do the same thing?

atiselsts, 2020-01-03:
> Maybe rpl-classic should do the same thing?

Maybe. However, that is a different PR that would be quite a big change in RPL Classic. For now I'm interested in getting #1165 merged as that one fixes the behavior of link stats under the current behavior of RPL classic.

See the bigger issue about consistency between neighbor tables, by the way:  #1054

PR #1165 (2020-01-03): avoid adding link stats back to neighbors that are dead
This should greatly improve network rebuilding time when the root node is rebooted or replaced.

Fixes #938
closed 2020-01-03 by simonduq