contiki-ng/contiki-ng #939: ways to Send the data from host to border router only

PranavD10, 2019-05-01:
Hi,
I want to ask if there is any way I can pass the information to the Border router.
I have used the net-uart program and made this to work as a border router. But when I started tunslip6 it shows the message that the prefix has not been assigned.
Why so
one ways I think of is because netstack has not set the prefix and the there is different netstack for node and border router.

Thank you

yatch, 2019-05-01:
If you followed [this document](https://github.com/contiki-ng/contiki-ng/tree/develop/examples/platform-specific/cc26x0-cc13x0/cc26x0-web-demo) and encountered an issue, please describe the issue precisely.

Regarding `tunslip6`, [this wiki](https://github.com/contiki-ng/contiki-ng/wiki/Tutorial:-RPL-border-router) may help.

closed 2019-05-01 by yatch

PranavD10, 2019-05-02:
Hi,
I followed the said document.
Here is the things I tried 
I first created the border router in the example and created the nodes with web-demo example.
Here I can send and receive the data to the nodes on the given port from host connected to border router and running tunslip6.
After that I had some custom command messages I want to send from the host to border router. So I use the net-uart example with the border router and uploaded the firmware in the hardware. I observed that the border router is not setting the prefix.
I think its because the tcpip event and the handler in the netuart is not allowing the packets to be proceed further...

Thank you

PranavD10, 2019-05-02:
Hey,
I figured it out.
It is working just I had to disable the uart the web-demo and its working fine.
Thank you