contiki-ng/contiki-ng #949: TypeError during script run

abhiverma866, 2019-05-06:
While trying to run the below script i am getting a type error. Possible solutions ??

```
TIMEOUT(300000, log.log("Performance Calculation" + "\n"));

packetsReceived= new Array();
packetsSent = new Array();
timeReceived = new Array();
timeSent = new Array();
count = new Array();
nodeCount = 10;
//data_length = 23;
senderID = 0;
receiverID = 0;
PDR=0;
//e_count = 0;

for(i = 1; i <= nodeCount; i++){
    packetsReceived[i] = 0;
    packetsSent[i] = 0;
    timeSent[i] = 0;
    timeReceived[i] = 0;
}

/*
DATA send to 1 'Hello 1'
Data is Hello 1 from the client
DATA recv 'Hello 1 from the client' from 5
*/

while(1){
    YIELD();
    msgArray = msg.split(' ');
    if(msgArray[1].equals("send")){  
        if(msgArray.length == 6){
            // sent packet
            senderID = parseInt(msgArray[3]);
            packetsSent[senderID]++;
            timeSent[senderID] = time;
        }
    }
    if(msgArray[1].equals("recv")){
        if(msgArray.length == 9){
            receiverID = parseInt(msgArray[8]);
            packetsReceived[receiverID]++;
            timeReceived[receiverID] = time;
            log.log("receiverID " + receiverID + " PacketReceived= " +packetsReceived[receiverID] + "\n");
                if (timeReceived[receiverID] > 0){
                    count[receiverID]++
                }       
            totalReceived = totalSent = 0;
            totalclient=0;
            for(i = 1; i <= nodeCount; i++){
                totalclient++;
                totalReceived += packetsReceived[i];
                totalSent += packetsSent[i];
                log.log("MoteID= " + i + " ReceivedPackets= " + packetsReceived[i] + " SendingPackets= " + packetsSent[i] + "\n");
            }

            log.log(" Generated Packets " + totalSent + "\n");
            log.log(" ReceivedPackets " + totalReceived + "\n");
            PDR=(totalReceived / totalSent) * 100
            log.log("Packet Delivery Ratio" + PDR + "\n");
        }
    }   
}
```

Java stack trace:
<eval>:39 TypeError: Cannot read property "equals" from undefined
	at jdk.nashorn.internal.runtime.ECMAErrors.error(ECMAErrors.java:57)
	at jdk.nashorn.internal.runtime.ECMAErrors.typeError(ECMAErrors.java:213)
	at jdk.nashorn.internal.runtime.ECMAErrors.typeError(ECMAErrors.java:185)
	at jdk.nashorn.internal.runtime.ECMAErrors.typeError(ECMAErrors.java:172)
	at jdk.nashorn.internal.runtime.Undefined.get(Undefined.java:157)
	at jdk.nashorn.internal.scripts.Script$Recompilation$17$40$\^eval\_.run(<eval>:39)
	at jdk.nashorn.javaadapters.java.lang.Runnable.run(Unknown Source)
	at org.contikios.cooja.plugins.LogScriptEngine$4.run(LogScriptEngine.java:300)
	at java.lang.Thread.run(Thread.java:748)

atiselsts, 2019-05-06:
Hi Abishek,

after you execute  `msgArray = msg.split(' ');`, the length of `msgArray` may be less than 2 (if the string does not contain any spaces). In that case `msgArray[1]` will be undefined.

In general, this type of questions are better suited for Stack Overflow (https://stackoverflow.com/questions/tagged/contiki) or other help channels.

closed 2019-05-06 by atiselsts

mhmdnaufal, 2019-06-01:
hey what is the solution of this issue? what variable that i need to change to run the script?

abhiverma866, 2019-06-01:
Check length of the string before extracting particular index. In this way you can avoid looking for non existing index.

mhmdnaufal, 2019-06-01:
which line i should change? can you help me?

abhiverma866, 2019-06-01:
if(msgArray[1].equals("send"))
{ if(msgArray.length == 6)
{ // sent packet 
_______________________________________

if(msgArray[1].equals("recv"))
{ 
if(msgArray.length == 9)
{

just change the order of these lines...

mhmdnaufal, 2019-06-01:
like this? 

while(1){
    YIELD();
    msgArray = msg.split(' ');
    if(msgArray[1].equals("recv")){
        if(msgArray.length == 9){
            receiverID = parseInt(msgArray[8]);
            packetsReceived[receiverID]++;
            timeReceived[receiverID] = time;
            log.log("receiverID " + receiverID + " PacketReceived= " +packetsReceived[receiverID] + "\n");
                if (timeReceived[receiverID] > 0){
                    count[receiverID]++
                }    
    if(msgArray[1].equals("send")){  
        if(msgArray.length == 6){
            // sent packet
            senderID = parseInt(msgArray[3]);
            packetsSent[senderID]++;
            timeSent[senderID] = time;
       }

did you mean like this? i still get same error

abhiverma866, 2019-06-01:
if(msgArray.length == 6)
{
if(msgArray[1].equals("send"))
{

mhmdnaufal, 2019-06-01:
thank you very much sir may god bless you