contiki-ng/contiki-ng #961: Cannot run out-of-tree tests inside docker

g-oikonomou, 2019-05-12:
As reported in #897, it is currently impossible to test out-of-tree builds inside docker.

Those builds rely on on [contiki-ng/out-of-tree-tests](https://github.com/contiki-ng/out-of-tree-tests) being mounted at `$(HOME)/out-of-tree-tests`, but in our current docker image it is not: It only gets mounted inside travis by `.travis.yml`.

simonduq, 2019-10-11:
Yes it does sound like a `contikier` issue: you need a virtual mount, something like: `-v $CNG_PATH:/home/user/contiki-ng/out-of-tree-tests` (maybe you also need `--mount type=bind,source=... destination=...`

g-oikonomou, 2019-10-12:
https://github.com/contiki-ng/contiki-ng/wiki/Docker#running-out-of-tree-tests

closed 2019-10-12 by g-oikonomou