contiki-ng/contiki-ng #968: Context Aware Next Gen Firwall | Contiki OS

kaleemullah360, 2019-05-21:
An Idea to context aware firewall in Contiki OS

**Key features**

**Sophisticated application and web control**

» Deep Packet Inspection (DPI) firewall
» Application control
» Web control
» URL filtering

**Comprehensive threat protection**

» Intrusion Detection/Prevention System (IDS/IPS)
» IP reputation services
» Malware protection
» Anti-virus

**Advanced user connectivity**

» Secure remote VPN access
» Site-to-site VPN connectivity
» Advanced routing capabilities
» Resilient WAN connectivity

**Powerful centralized management**

» Management and monitoring
» Allied Telesis Management FrameworkTM (AMF)
» Flexible licensing options

content copied from [Source](https://www.alliedtelesis.com/sites/default/files/ngfw_features_overview_reve.pdf)

g-oikonomou, 2019-05-21:
Spam. I have removed the link to PDF with commercial material and closing.

kaleemullah360, 2019-05-22:
Not spam, I was studying next gen firewalls. and as I am using contiki since 2015, that why I copied a new idea for contiki over here

simonduq, 2019-05-23:
Closing this, please use Gitter for general open discussions, Github issues for bug reports and issues.
https://github.com/contiki-ng/contiki-ng/wiki/Where-to-report-issues,-ask-questions,-etc.%3F

closed 2019-05-23 by simonduq