contiki-ng/contiki-ng #971: change zoul RF from 2.4G to 868MHz

prodomo, 2019-05-24:
Hi
I'm running a test with 11 zoul (1 Root, 10 client).
I use spectrum analyzer to see real tx signal,
I only saw signal on 2.4GHz.
What should I do to set zoul RF from 2.4GHz to 868MHz.

g-oikonomou, 2019-05-24:
I started looking at an answer to this for you, and I stumbled across a bug.

You will need to:

* `#define NETSTACK_CONF_RADIO           cc1200_driver`
* Have a look in `arch/platform/zoul/contiki-conf.h`. Near line 118 there is a bunch of other defines.

Unfortunately if you were to make the required changes we fail to compile.

I'll open a PR soon, including travis tests to make sure we don't have a regression.

Flagging a bug here for now, since this switch is currently impossible.

prodomo, 2019-05-28:
I changed config as you say and the RF has been changed to 868MHz.
I used example/rpl-udp, server and client can rev DIS, DIO each other, 
but "if(NETSTACK_ROUTING.node_is_reachable() && NETSTACK_ROUTING.get_root_ipaddr(&dest_ipaddr))" always be false in client.
Application layer always can't get dag root .

closed 2019-07-02 by joakimeriksson