contiki-ng/contiki-ng #977: TSCH locks up on setting channel on Simplelink-cc13xx-cc26xx platform

hseokman, 2019-05-28:
While I was looking into implementing TSCH on Simplelink platform, I found an issue.

set_value with RADIO_PARAM_CHANNEL calls set_channel function which calls netstack_sched_fs.
netstack_sched_fs uses RF_runCmd which is synchronous function that should not be called from the interrupt handler. tsch_slot_operation uses rtimer which calls the function from the interrupt handler. When tsch_slot_operation calls to change channel, the device locks up.

Looking at cc26x0-cc13x0 implementation, set_channel function does not send any radio command. The channel was set when the radio is turned on.

hseokman, 2019-05-29:
Since the function name is netstack_sched_fs, shouldn't it schedule the command? Shouldn't it call RF_scheduleCmd instead of RF_runCmd?

hseokman, 2019-05-29:
The same issue is present in netstack_sched_ieee_tx and netstack_sched_prop_tx. They schedule to transmit a packet and wait for completion with RF_pendCmd which will block from the interrupt handler of rtimer.

zhuqiyun, 2019-07-19:
focus

g-oikonomou, 2019-08-31:
#1038

closed 2019-08-31 by g-oikonomou