contiki-ng/contiki-ng #981: Last node in the dag

PranavD10, 2019-06-03:
Hey,
How can I find the last element in the DAG if I use RPL-lite?
Thank you

debug-ito, 2019-06-05:
What do you mean by "the last node"? A leaf node, i.e., a node with no children ??

PranavD10, 2019-06-13:
I want to find the DAG structure the nodes are connected...how will I know that...???

debug-ito, 2019-06-13:
If you use rpl-lite, `rpl_neighbor_print_list` function (os/net/routing/rpl-lite/rpl-neighbor.h) prints the list of RPL neighbors, which includes the preferred parent node. By combining that information from all nodes in the DAG, you can construct the upward routes (DAG). In addition, calling `rpl_dag_root_print_links` (os/net/routing/rpl-lite/rpl-dag-root.h) in the root node will show all entries in the source routing table. That corresponds to the downward routes.

closed 2019-10-04 by g-oikonomou