contiki-ng/contiki-ng #983: Build issue when using sensors_sensor structure in sensor.h

ahmedmibrahim, 2019-06-13:
Hello, 
I found a build problem whenever the code uses the sensors_sensor data structure defined in os/dev/sensors.h. 
Below is a simple example that gives: "undefined reference to button_sensor" error,:

`
#include "contiki.h"
#include "dev/leds.h"
#include "dev/button-sensor.h"
#include <stdio.h>

PROCESS(led_button_process, "Leds & Buttons process");
AUTOSTART_PROCESSES(&led_button_process);
PROCESS_THREAD(led_button_process, ev, data)
{
  PROCESS_BEGIN();
  SENSORS_ACTIVATE(button_sensor);
  while(1) {
    printf("Press the user button\n");
    PROCESS_WAIT_EVENT_UNTIL(ev == sensors_event && data == &button_sensor);
    leds_toggle(LEDS_GREEN);
    printf("The sensor is: %u\n", leds_get());
  }
  PROCESS_END();
}
`

My make file is:

`
CONTIKI_PROJECT = button_sensing
TARGET=cc26x0-cc13x0
BOARD=launchpad/cc1350

CFLAGS += -g
all: $(CONTIKI_PROJECT)

CONTIKI = ../..
include $(CONTIKI)/Makefile.include
`

g-oikonomou, 2019-06-13:
The CC1350 launchpad uses the new button API: Have a look at `examples/dev/button-hal/` to see how to use it. `os/dev/button-hal.h` for detailed API documentation.

closed 2019-06-13 by g-oikonomou

ahmedmibrahim, 2019-06-13:
Thanks for the quick answer, but I think the problem is not with the api, I tried other sensors with the launchpad and it fails in the same manner, take for example the cc26xx-web-demo, I tried to use any of the sensortag board sensors with the launchpad and I arrived at the same error: "undefined reference to ****_sensor"

Do you know how to use, for example, the hdc_1000_sensor in the cc26xx-web-demo with the launchpad?

g-oikonomou, 2019-06-13:
Does your launchpad have an HDC1000?

ahmedmibrahim, 2019-06-13:
It has a very similar sensor (htd_21d), the sensor is not an issue, the issue is build crashes.
Are you able to build the project on the launchpad, while porting "any" of the sensortag sensors?

g-oikonomou, 2019-06-13:
> Are you able to build the project on the launchpad, while porting "any" of the sensortag sensors?

Yes. One needs to modify the build system such that the sensor driver gets compiled.

ahmedmibrahim, 2019-06-14:
Hi I figured out the error , I was missing something in the make file, details are here
https://e2e.ti.com/support/wireless-connectivity/sub-1-ghz/f/156/p/809307/3000233#jktype=e2e&pi320995=3

Thanks for your help!