contiki-ng/contiki-ng #986: Jumping into a statement expression with a computed goto has undefined behavior

LDVG, 2019-06-18:
I was looking up the `lc-addrlabels.h` implementation and saw that the `LC_SET(s)` macro uses a statement expression within its `do ... while(0)` construct:

https://github.com/contiki-ng/contiki-ng/blob/beb623825f26da1ee234ccffe4e344beed62100f/os/sys/lc-addrlabels.h#L75-L76

According to the [GCC documentation on statement expressions](https://gcc.gnu.org/onlinedocs/gcc-8.2.0/gcc/Statement-Exprs.html), jumping to this label would have undefined behavior (emphasis mine):

> Jumping into a statement expression with goto or using a switch statement outside the statement expression with a case or default label inside the statement expression is not permitted. **Jumping into a statement expression with a computed goto (see [Labels as Values](https://gcc.gnu.org/onlinedocs/gcc-8.2.0/gcc/Labels-as-Values.html#Labels-as-Values)) has undefined behavior.**

Should the statement expression in `lc-addrlabels.h` be removed or am I missing something? What's the rationale for using it there?

debug-ito, 2019-06-24:
As far as I read in GCC documentation, I think you are right. I'm not sure why we use a statement expression here.

Personally I use `lc-switch.h` for LC implementation, which is the default in Contiki-NG. I suspect `lc-addrlabels` is not so extensively tested.

LDVG, 2019-06-24:
Right. I guess just removing it, like so

```c
#define LC_SET(s)                               \ 
  do { __label__ resume; resume: (s) = &&resume; } while(0) 
```
will not break anything since

- The operand `s` is only evaluated once (important if `s` would be a function call with side effects)
- The multi-statement operations in the macro are handled by the `do { ... } while(0)` construct
- The declaration of `resume` is still in the beginning of the block and scoped by the construct
- The macro is not depending on statement expressions ability to "return" values

unless it's there for some other reason. :thinking:

commit 92202318 (2019-06-27): lc-addrlabels: remove statement expression (fixes #986)

According to GCC documentation, jumping into a statement expression with
a computed goto (using labels as values) has undefined behavior and
should be avoided.  However, this is exactly what is happening in the
current LC_SET implementation.

Fortunately, the statement expression which contains the label in LC_SET
can safely be removed since the operand is only evaluated once, and
multi-statement operations of the macro are already handled by a
do-while(0) construct.
commit b85de1c9 (2019-06-27): Merge pull request #990 from LDVG/develop

lc-addrlabels: remove statement expression (fixes #986)
closed 2019-06-27 by g-oikonomou